cleanflight/cleanflight #2720: Prepare for 2.0.1 release

hydra, 2017-04-14:


emma0223, 2017-04-14:
I have a spracingmini f3 with a sp racing osd/pdb and my quad became a paper weight since the new cf gui 2.0 update. I can't download gui 1.2.4 anywhere to mod my osd and with v2.0 on fc and pbd osd I am unable to do anything please fix this

neverknownbetter, 2017-04-14:
I have a sp racing f3 deluxe FC. Can not flash new 2.0 firmware to FC. Will only except last version before 1.14.2. Is there any fix for this as I want to piggy back on a sp racing f3 OSD?
Also unable to flash new firmware on sp racing f3 OSD as well. STM 32 virtual COM port installed and recognize in device manager. But getting failed serial port to open.. please help...

emma0223, 2017-04-14:
@neverknowbetter if I'm rigth there's now hex file supporting the sp f3 Delux as a master for a slave sp osd pdb... the only available hex are for the f3 and f4 Evo .... I have the same problem I have a spf3mini and can't use my sp pdb osd with the new updates... the only fix is to wait for the developers to make new releases o for you to modified the hex file i

jimbaxter, 2017-04-14:
I am using a NAZE32 Full Rev 5 board with the FrSky D8R-II plus which is in Parallel PWM mode and a GPS receiver attached.
The Throttle and Rudder are being incorrectly interpreted as AUX1 and AUX2 channels instead of the Throttle and Rudder.
Oddly the Ailerons and Elevators are ignored completely, when they should still work correctly as they are before the RX/TX pins 3 and 4.
Initially it appeared that the channels were not being offset after the GPS pins 3 and 4, but with the Ailerons and Elevators not working I feel it is more complicated then that.

hydra, 2017-04-15:
@neverknownbetter Yes, you are correct, the SPRacingF3 target needs to have OSD compiled in, we will fix that.  Thanks for the bug report.

hydra, 2017-04-15:
@jimbaxter thanks for the report.  Can you provide 
a) a photo showing how you have your channels connected to the FC
b) a config dump

jimbaxter, 2017-04-15:
@hydra I will get a photo, though the connections have not changed since the previous working Cleanflight 1.14. It is connected according to the specs in http://www.abusemark.com/downloads/naze32_rev3.pdf

I have attached the cleanflight dump for the previous 1.14 working version and the new non-working 2.0 version in the following zip file.

[cleanflight_dump.zip](https://github.com/cleanflight/cleanflight/files/923467/cleanflight_dump.zip)

neverknownbetter, 2017-04-15:
When the firmware is fixed for both boards (Sp racing f3 deluxe and sp racing osd). Is it safe to flash both boards when they are connected or do them separately?

hydra, 2017-04-15:
@neverknownbetter disconnect both boards and unplug battery.  flash each board via usb, one at a time.

Gravity-Thief, 2017-04-15:
Running a DODO FC. Put 2.0.0 on yesterday.
Now my quads **motors do NOT spin when armed, even though I have that option selected.**

luxflux, 2017-04-15:
My configuration which works on Cleanflight 1.14 doesn't work anymore after an upgrade to cleanflight 2.0. It cannot establish a connection to the motors/escs, they constantly beep every second. Same wiring works with Cleanflight 1.14.

What do you need from me to debug this?

McGiverGim, 2017-04-16:
@hydra I've a similar problem that @luxflux, @Gravity-Thief, etc. With the latest master, something is wrong with ESC/Motors. For example, if I go to CLI and type `tasks`... the motors start to spinning!
I cannot arm it (my CPU usage is under 20% in my NAZE32), the ESC enter in mode programming when I go to motors tab and try to test them, etc.
You know that I have been testing 2.0.0 since the beginning, and this problems are new.

pawelsky, 2017-04-16:
PPM RSSI does not work anymore in 2.0.0 (#2724)

McGiverGim, 2017-04-17:
@hydra I've been making more tests, my problem does not happen with version 2.0.0 "oficial", this only happens with latest master and/or my customized version. I will make more tests to isolate the commit/customization that causes the problem.

StraTech, 2017-04-17:
Sparky 2.0 LED Strip support
The LED strip feature is not currently supported .
Could you please Enable Led strip support for Sparky 2.0 - via PC8 or PB14 or PB15 
(I apologize if this isn't the correct place to request function)
Thankyou for including Sparky 2.0 in Cleanflight!

hydra, 2017-04-18:
@jimbaxter have you tried using the 'resource' cli command?

hydra, 2017-04-18:
@ Gravity-Thief can you provide a config dump + cli diff?  Have you tried resetting to defaults and re-testing?

Rogue-UAV, 2017-04-18:
Using SP Racing NEO stack. Although you can select different VTX channels using sticks, the channel and or band does not actually change. Appears to be stuck on whatever the default channel is. Also LED strip does not properly address LED's when changing the color, not too concerned with this though.

Gravity-Thief, 2017-04-18:
@hydra here's my dump and diff. Will try resetting to defaults and re-testing once im back from work and see if that does anything.

[Gravity-Thief diff.txt](https://github.com/cleanflight/cleanflight/files/928551/Gravity-Thief.diff.txt)
[Gravity-Thief dump.txt](https://github.com/cleanflight/cleanflight/files/928552/Gravity-Thief.dump.txt)

azmodie, 2017-04-18:
I have an eachine assault 180. all worked fine in cf 1.14 with map AETR1234. in cf 2.0 the channels over parallel_pwm from flysky-i6 transmiter is wrong. throttle and yaw are on aux 1 & 2 same as @jimbaxter. also rxfail on channels 0-3 is set to a. so the channels are in failsafe and do not responsd. setting the rxfail 0-3 to h allows me to see movement on sticks with map AE34TR12 although i still have no roll input.

StraTech, 2017-04-18:
Hi thanx, I have tried reset the settings to default & re-flashing firmware 2.0.0.
I have attached the requested Dumps, I have noticed  that some resource are mapped more than once to the same port ,
resource BEEPER 1 C09
resource PWM 5 C09
resource PPM 1 C07
resource PWM 1 C07
resource SERIAL_RX 6 C07
Also with 'resource list' it appears C07 is Free ?
I'm not sure if this normal or maybe part of the problem. 
I'm able to access the LED Strip tab in CONFIG but not able to assign any LED via Wiring order mode.
Also noticed that it is displaying '0 Remaing' Leds to assign. 
Thank u for ur support!
[CliDiffSpark2.txt](https://github.com/cleanflight/cleanflight/files/929763/CliDiffSpark2.txt)
[CliDumpSpark2.txt](https://github.com/cleanflight/cleanflight/files/929764/CliDumpSpark2.txt)
[CliResourceListSpark2.txt](https://github.com/cleanflight/cleanflight/files/929766/CliResourceListSpark2.txt)

StraTech, 2017-04-18:
I've been trying to remap resource's via cli but I'm unable too!
Tried :    'Resource LED_STRIP 1 B15'
              'Resource LED_STRIP B15'
No Luck!
I've noticed that by comparing the Target.c file's of SPRF3 & Spark 2.0 that LED_STRIP doesn't appear to be defined in the Sparky 2.0 Target.c 
Not sure if this is the fault as I haven't  programmed since the old Apple/Atari 6502 8bit days.

hydra, 2017-04-18:
@Rogue-UAV you need a new build to use sticks and button.  check here - https://github.com/cleanflight/cleanflight/pull/2721#issuecomment-294303424

hydra, 2017-04-18:
@StraTech led strip is not currently supported on the Sparky 2.  Do not try it, it won't work right now.

hydra, 2017-04-18:
@Gravity-Thief I have no issues with motor_stop here. works just fine.  reset to defaults and try again.  please provide steps to repeat/reproduce in a new issue and we will address once you've given us more details.

hydra, 2017-04-18:
@jimbaxter did you try using the resource command?  If not try that.  If you did and still have an issue please provide steps to repeat/reproduce in a new issue and we will address once you've given us more details.

hydra, 2017-04-18:
all known confirmed reproducible issues fixed, merged or deferred.  cf_patch_v2.0.1 branch and tag created.

closed 2017-04-18 by hydra

FreshRC, 2017-04-18:
Hail Hydra ;)
I have SP Racing F3 EVO.
With new CF v2.0.1 it has the same issue as I had with BF.
When using DSHOT I need to remap motor 4 to motor 5 resource and motor 5 pin as motor 4 will not spin without this workaround. 
But new issue I didn't have with BF is that now I have a conflict with resource for SoftSerial (1 or 2, TX or RX need to check as I don't remember from my head)

hydra, 2017-04-18:
v2.0.1 is out now.  further bug reports go on the v2.1.0 issue, here: https://github.com/cleanflight/cleanflight/issues/2742

(unless we need a patch release)

hydra, 2017-04-18:
@FreshRC your issue with remap of motor 4 to 5 is expected, a well known and documented hardware limitation.

Rogue-UAV, 2017-04-20:
@hydra Still having issues changing VTX channel on SP Racing NEO stack. I can change the channel using sticks or button with 2.0.1, shows it is on a different channel, but the channel does not actually change.

Bladeoutlaw, 2017-04-21:
hey i have naze with the newest update and in the receiver screen when i adjust my stick max middle and min. it wont save at all it keeps going to default kinda. not sure what to do and cant take a screen shot of it