cleanflight/cleanflight #2724: Cleanflight v2.0.0 RSSI issue.

Life-is-short, 2017-04-15:
Tested with SP racing F3, RSSI doesn't work with the latest cleanflight anymore, previous cleanflight version was working fine. It is impossible to see RSSI using receiver channel. Betaflight is working fine.

pawelsky, 2017-04-16:
Same on Naze32. Works on 1.14.2, does not work on 2.0.0.

Life-is-short, 2017-04-16:
Seems that newest cleanflight version has lots of bugs :( I will stick to betaflight, it has no such issues.

hydra, 2017-04-17:
@Life-is-short CF2 is based on BF.  If it's a bug in Cf then it'll be a bug in BF too.  Please keep issue reports free from such trolling/opinionated/incorrect comments and instead be as factual, on topic and helpful as possible.

Life-is-short, 2017-04-18:
Dear hydra,

which part of my comment is trolling/opinionated/incorrect. You didn't like that I told latest cleanflight has lots of bugs? Just check what people reports, also as I told before RSSI doesn't work, I tested in 6 models, after I flash Betaflight, it starts working like it should. Just let me know if I can help to debug RSSI issue somehow.

hydra, 2017-04-18:
Thanks, that reply is much more constructive.

I need more information about what you are using to locate the problem.

Please post a config dump of
A) your CF setup
B) your BF setup

apavel3458, 2017-04-18:
Hi there, I have a naze32 running cleanflight 2.0.0.

RSSI confirmed NOT working: 

Two screenshots:  
1.  Receiver section showing channel 8 input WORKING! (feeds RSSI/Link Quality)
2. When I look at the "status" section, shows RSSI = 0%
3.  Attached CLI settings. 

[image]
[image]

[cli_dump.txt](https://github.com/cleanflight/cleanflight/files/930832/cli_dump.txt)

hydra, 2017-04-19:
this PR is a good excuse for me to cleanup the RSSI mess in the codebase.  I'll take a look into it.

apavel3458, 2017-04-19:
Awesome, look forward to it!

hydra, 2017-04-20:
bug confirmed, the issue is only with RSSI via MSP in the configurator.

RSSI via OSD or via TELEMETRY works just fine.  The problem ocurred when the OSD Slave code was added.

hydra, 2017-04-20:
@Life-is-short btw, BF master will also have this bug, so your statement about BF having less issues is proven incorrect. Here's a link to the code in Betaflight for the record: https://github.com/betaflight/betaflight/blob/master/src/main/fc/fc_msp.c#L600

Just thought you'd like to know.

Life-is-short, 2017-04-21:
Thanx, strange, but RSSI is working fine with Betaflight. Waiting for the fix so I can use Cleanflight again :) Thank you for the great job you all guys do!

hydra, 2017-04-22:
will merge this fix into 2.0.2

closed 2017-04-22 by hydra