cleanflight/cleanflight #2725: Cleanflight - Configurator 2.0.0 can not use the BLHeliSuite connection problem

E120103016, 2017-04-15:
Today, the use of Configurator 2.0.0 and v2.0.0 program set up. Then BLHeliSuite set ESC. Can not use. 2.0.0 program is not set BLHeli Bootloader (cleanfight). How to solve this problem. Trouble Thank ...

McGiverGim, 2017-04-15:
Update blheli suite to the latest version.

stanislavdimov, 2017-04-15:
Cleanflight - Configurator new update can not use CF old ver /ver.1***/ with NAZE32 ver 6  /flash but not work/!?!?!?! Work only with new CF ver 2.0! But in new version CF do not have more GPS!
Cleanflight - Configurator old version no more work!
What we happy MrGiverGim?

E120103016, 2017-04-16:
Today, BLHeli can be updated. Thank you very much the two predecessors. Thank you again!

closed 2017-04-17 by E120103016