cleanflight/cleanflight #2726: Smart Port Telemetry

SavvyZMR, 2017-04-15:
I have a Naze32 w/ a fr-sky X8R running sbus connection w/ telemetry running on smart port soft serial 2 w/ a Taranis QX7  also I've: "set tlm_inversion = on" Post cleanflight 2.0 update. I have this exact same setup running on two different quads and both quads have done the exact same thing with two seperate recievers and two seperate transmitters. Basically pre cleanflight 2.0 update all sensors worked fine post cleanflight 2.0 update only RSSI Voltage and RSSI Range sensors are working/ discoverable the sbus is running on Uart2 -serialRX on pins -5V, +5V and pin 4 telemetry on pins 5,6 with a Y-Cable to the smart port signal pin

Reconnec7ed, 2017-04-15:
Also experiencing a similar issue

Steele450, 2017-04-16:
I have a very similar situation as to SavvyZMR, where my telemetry is not working in cleanflight 2.0. The only sensors that I can receive with my taranis qx7 are RSSI voltage and RSSI range. My receiver is a frsky x8r running with sbus and  my fc as a naze32 rev6

kcunreal, 2017-04-16:
Same here, but using Frsky X4R RX. No VBAT or current telemetry available. Switched back to 1.14.2 and everything works as normal.

hydra, 2017-04-17:
please provide a config dump.  certainly it works fine on the F3 and F4 targets in CF2.x.

can you also check the status of the telemetry inversion setting 'tlm_inversion' perhaps it is wrong?

Arbitrator, 2017-04-18:
I just tested a X4RSB with a Naze32 (full, rev6) and it is sending telemetry to my Taranis. I connected via SBUS and SmartPort on 5/6 (SoftSerial1) with 'tlm_inversion = ON' as well as SoftSerial and Telemetry enabled.

hydra, 2017-04-18:
works fine here, used softserial 2 on a naze32 with an X8R.  Used 'set tlm_inversion=ON', enabled telemetry feature. connected diode as per manual (same as HoTT) and it works fine.

Using CF 2.0.0.
```
defaults
```
then
```
feature SOFTSERIAL
feature TELEMETRY
serial 31 32 115200 57600 0 115200
set tlm_inversion = ON
```

closed 2017-04-18 by hydra

kcunreal, 2017-04-18:
Before Cleanflight 2.0 everything work fine. Now I cannot receive current telemetry. I am using Acro Naze32 and X4R-II. Vbatt works fine. Everything is wired correctly, no damaged components, and as far as I can tell everything is configured as it is supposed to be (because after all it worked pre Cleanflight 2.0).  I was told I had wired something wrong and that's why it's not working, which is obviously not the case.

SavvyZMR, 2017-04-19:
I was finally able to get mine working with the x8r I had to turn off MSP on softserial

SavvyZMR, 2017-04-19:
also I originally thought that the problem was with cleanflight 2.0 it wasn't I was able to reinstall cleanflight 1.2.4 along side cleanflight 2.0 and I had the same problem and before I realized that msp had to be off on softserial I was able to get it working by arming the quad during the sensor discovery on the taranis and the sensors were discovered immediately but once I turned off msp the cpu load on the naze went down by half and all the sensors work fine including Vfas on the X8R anyone interested in reinstalling ver 1.2.4 along side 2.0 there is a painless360 video on youtube showing you how it was really easy