cleanflight/cleanflight #2728: Servo-Tilt not working on naze32 rev6 acro/6dof

cantaxONE, 2017-04-16:
After activating servo-tilt in the configuration tab, motor connectors does not switch from 1-4 to 3-6.
I tried to connect the servo to port 5 or 6. Doesn't work.
I got the same problem with Betaflight 3.1.6.
On Cleanflight 1.14.2. it worked as it should.

cantaxONE, 2017-04-16:
Ok, I solved it.
The problem seems to be, that after i turn on "Servo Tilt", the Motor/Servo Ports are not changed. They just stay the same as if "Servo-Tilt" isnt enabled.
This is how i got it working through cli:

Ports before and after enabling "Servo-Tilt"
resource MOTOR 1 A08
resource MOTOR 2 A11
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B08
resource MOTOR 6 B09
resource SERVO 1 NONE
resource SERVO 2 NONE

I changed it manually to these values:
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource SERVO 1 A08
resource SERVO 2 A11

I did manually what cleanflight 1.x did automatically.
I think this will be quite simple to fix.

hydra, 2017-04-17:
yes, the documentation could do with updating.  can you create a PR to update it?

hydra, 2017-04-18:
closing this issue, please create a documentation PR when you have time.

closed 2017-04-18 by hydra