cleanflight/cleanflight #2735: Link for "Documentation For 2.0.0" does not work properly.

Arbitrator, 2017-04-18:
It currently opens the browser to 'https://github.com/cleanflight/cleanflight/tree/v2.0.0/docs', with a 404 message 'This is not the web page you are looking for.'

stanislavdimov, 2017-04-18:
Perhaps line in Documentation for Building in Eclipse.md
"Set your build and debug targets by going to project properties -> C/C++ Build and choose the Behaviour tab. Replace "all" in the build box with "TARGET=xxx DEBUG=GDB" where xxx is your platform such as NAZE"
change with
"Set your build and debug targets by going to project properties -> C/C++ Build and choose the Behaviour tab. Replace "all" in the build box with "make TARGET=xxx DEBUG=GDB" where xxx is your platform such as NAZE"
or not?

hydra, 2017-04-18:
Thaks for the report. I will take a look.

hydra, 2017-04-19:
fixed in latest configurator master - wait for CF configurator v2.0.1.

closed 2017-04-19 by hydra