cleanflight/cleanflight #2737: [DOCS, OSD/PDB] - Please provide documentation for bus switch functionality

biank88, 2017-04-18:
In my current setup:
- flight controller: SPF3
- osd: SPF3-OSD-PDB
I cannot connect (via the configurator) to the fc-USB if the osd/pdb is powered. This behaviour would be expected since the SPF3 UART1 is shared between the osd headers and the USB Silabs chip... However, since there is a `bus switch` on the osb/pdb, I wonder weather there is something I'm missing.

[image]

hydra, 2017-04-18:
The behavior is not erratic, it's consistent when 2 devices try and use the same serial signals.

the SPRACINGF3 cannot be powered at the same time as the SPRACINGF3OSD/PDB unless the bus switch is disabled on the SPRACINGF3.

There is currently no software support for the bus switch other than to enable it at boot on the OSD/PDB board.

At one point it was envisaged that there might be a need to disable the bus switch. There are some flashing scenarios where the bus switch helps too.

The SPRACINGF3EVO and SPRACINGF4EVO don't use the CP2102 that the SPRACINGF3 has and thus do not have the issue.

There's no way the SPRACINGF3 can detect USB connection, so no way to disable to bus switch.
It's technically possible to detect USB connection on the OSD to disable the bus switch but nothing like that has been implemented.

Perhaps it's possible to disable the bus switch via some GUI control so that the FC can be configured.

This only affects the SPRACINGF3.  SPRACINGF3EVO, SPRACINGF4EVO are unaffected.  SPRACINGF3 has been discontinued for over a year now.

If you have any ideas please let me know, perhaps we could add something that might help but I can't think what/how at this time.

closed 2017-04-18 by hydra

biank88, 2017-04-19:
Thanks! I think you should mention this problem in the docs for the SPF3-OSD/PDB.
In my case, in order to maintain the possibility to program the FC, I've desoldered the uart headers and I've rewired the OSD/PDB-UART1 to SPF3-FC-UART2.