cleanflight/cleanflight #2741: Stick inputs not showing up in cleanflight

ripley123, 2017-04-18:
I'm running a lemon rx satellite receiver and a spektrum DX6 as well as an SPF3. I was able to bind the DX6 to the satellite using the cleanflight CLI however none of my stick inputs show up under the receiver tab.

I used: set spektrum_sat_bind=7 (DSMX 1024 22ms)

[image]
[image]

hydra, 2017-04-18:
please use forums for support, probably a configuration issue.  we're not aware of any issues with spektrum rx.

hydra, 2017-04-20:
another user just posted an issue (#2746) stating that they can see receiver input from a spektrum rx and an SPRacingF3 and had no issues seeing stick movements on the RX tab.  likely you just have a configuration issue.  please use forums for support, if you still have an issue please open a new issue and reference this one.

closed 2017-04-20 by hydra