cleanflight/cleanflight #2742: Prepare for v2.1.0 release.

hydra, 2017-04-18:


Gravity-Thief, 2017-04-19:
Running 2.0.1 now
Motors still do not spin at set minimum speed when armed.
Iv also noticed that the quad does not automatically un-arm after the set time.

[Gravity-Thief diff 2.0.1.txt](https://github.com/cleanflight/cleanflight/files/936063/Gravity-Thief.diff.2.0.1.txt)
[Gravity-Thief dump 2.0.1.txt](https://github.com/cleanflight/cleanflight/files/936062/Gravity-Thief.dump.2.0.1.txt)

McGiverGim, 2017-04-20:
@Gravity-Thief as I see in your diff, you have enabled MOTOR_STOP. If I'm not wrong you must disable it to achieve what you want.

McGiverGim, 2017-04-22:
@hydra I don't know if this is a problem of my hardware, but in the betaflight chat I've seen others with the same problem. I will return to CF1.x to confirm if this solves the problem or not.
With CF2.x sometimes the quad enter into failsafe mode randomly, but when I tried to remove some filters to get better performance, the problem start happen constantly. In this way is unflayable. 
I will try to make more tests but as I say there are other with the same problem but maybe the origin is different.
You can see my blacbbox here https://drive.google.com/file/d/0B6cSJRBCexb2LTIwQnJ0LWhRUFk/view?usp=drivesdk
Take a look to the second and third record, the quad was hovering at 10-15 meters of me only.

Gravity-Thief, 2017-04-22:
@McGiverGim - Makes no difference if I have MOTOR_STOP enabled or not. The problem still persists.

@hydra - Just tried 2.0.2 and still have the same issues.

hydra, 2017-04-22:
@Gravity-Thief please use forums for support.  Likely your hardware is configured wrong or your expectations are incorrect.

hydra, 2017-04-22:
@McGiverGim Please give me links to posts where others have the same issue.  Please also remind me which FC you are using.  Certainly I've been flying CF 2 on a variety of quads on different FC's with different RX systems and have not encountered any such issues.  Please gather some more information and take that topic to a new issue.  Please comment with a link here once you have done so.

Gravity-Thief, 2017-04-22:
@hydra - I am pointing out an issue. I am not asking for support. **Why is it if I go back to any firmware pre 2.0.0 I do not have this issue?** Everything is configured exactly the same that I can assure you. The only difference is the firmware.

McGiverGim, 2017-04-22:
@hydra I will do more tests tomorrow to be sure that is not a problem of my hardware (NAZE32) and then I will put you all this information. I suspect something is wrong with my serial rx cable. Thanks for your interest!

jimbaxter, 2017-04-22:
@hydra Sorry for the delay I have only just got round to investigating this issue again.
I have the issue:

> I am using a NAZE32 Full Rev 5 board with the FrSky D8R-II plus which is in Parallel PWM mode and a GPS receiver attached.
The Throttle and Rudder are being incorrectly interpreted as AUX1 and AUX2 channels instead of the Throttle and Rudder.
Oddly the Ailerons and Elevators are ignored completely, when they should still work correctly as they are before the RX/TX pins 3 and 4.
Initially it appeared that the channels were not being offset after the GPS pins 3 and 4, but with the Ailerons and Elevators not working I feel it is more complicated then that.

I have tested with 2.0.2 and it still has the issue.
This is the output of resource:
```
# resource
resource BEEPER 1 A12
resource MOTOR 1 A08
resource MOTOR 2 A11
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B08
resource MOTOR 6 B09
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 A02
resource PWM 4 A03
resource PWM 5 A06
resource PWM 6 A07
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A06
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
```

hydra, 2017-04-23:
Make some assignment changes using the resource command, don't just show us the output again :). We didn't change anything related since you last tried.

hydra, 2017-04-23:
@gravity-thief. Because lots of things changed!  Change your expectations of what you think the firmware is doing and make it do what you want it to.  I.e. check your settings vs what the firmware is configured to do.

hydra, 2017-04-23:
Again, please use forums for support, read documentations, wiki, tutorials watch videos, etc. Once you understand your issue you can either make adjustments to your equipment or the firmware. If then it STILL doesn't work let us know.

Gravity-Thief, 2017-04-23:
@hydra - **I must apologize, you are correct there is no issue.** I looked at a dump I had from a pre 2.0.0 firmware to see if I could see any differences, and I noticed that the min throttle setting before 2.0.0 was 1150 and from 2.0.0 it has changed to 1070. After changing back to 1150 all works as normal and motors once again spin up when armed. Thanks.

McGiverGim, 2017-04-23:
@hydra I'm not sure if I have found my problem or not, but I've found that I've too much resources assigned to SERIAL_RX (the two by defect and one more). Maybe this makes that something gets wrong and enters in failsafe mode randomly. I've removed the extra assignment and tomorrow I will test if the problem is gone or not. I don't know when was this resource assigned (maybe in a change of versions, I didn't do it manually).
But searching the problem, I've found two little bugs, nothing important, but here you have, if you want I can create a new issue for each of them:
- The first, the telemetry for iBus sends very few values to the TX. One of them is the temperature. I receive 0.0 as temperature. I've tested version 1.x of cleanflight and the temperature arrives ok. The other telemetry values arrive without problem. I think that was @unitware who made this, maybe he can tell us something ;). 
- The second is that if I disable telemetry into the config page of configurator, but I have telemetry iBus selected in the ports page (I use the same UART for Serial and Telemetry), the telemetry is sent ignoring the config value. I don't know if this is the expected behaviour.

mikeller, 2017-04-24:
@McGiverGim: The answer to your second question is that with the way that iBus RX and telemetry are implemented to run on the same port, iBus telemetry is not controlled by the `TELEMETRY` feature, so this is the expected behaviour.

McGiverGim, 2017-04-24:
Thanks @mikeller . And for the first question, seeing the code. It seemed that version 1.x showed the barometer temperature, and version 2.x shows the ESC temperature. My ESC didn't give this information. I don't know if the solution is to detect it and show one temperature or the other, or let it as is.

hydra, 2017-04-24:
@Gravity-Thief thanks for the update

hydra, 2017-04-24:
@McGiverGim remind me which exact rx/tx combo are you using again please.

McGiverGim, 2017-04-24:
@hydra I've a FlySky FS-i6s, and a FlySky FS-IA10B. Forgot my problems with failsafe mode by the moment, I'm doing tests. As it happens randomly I will need some days to be sure where is the problem. Thanks!

hydra, 2017-07-10:
closing this, 2.1.0-RC1 is out now.

closed 2017-07-10 by hydra