cleanflight/cleanflight #2745: Hott telemetry problem (GR-24, Mx-20) on SP F4 Evo

manjoel, 2017-04-20:
Hallo, i have an seriously pro f4 evo board, an GR-24 Hott receiver and an Mx-20 transmitter. On der Fctrl I have 2.0.1
Im connected over UART 2 for the receiver an the 4th pin is UART 5 by soldering the Pad TLM. In the configuration UART 2 = Serial RX and UART 5 = Telemetry output HOTT. I configuration tab telemetry is activated. I tested with Softserial and witout, but no telemetry.

I updated Mx-20 and Gr-24 firmware yesterday.

What could be the problem.

hydra, 2017-04-20:
No need for soft serial.

Double check the status of telemetry inversion setting, that's probably the issue, the default is ON but HoTT isn't inverted.  Try this in the CLI.

```
set tlm_inversion=OFF
save
```

Please report back ASAP.  

Certainly I have not updated the firmware on either the MX20 or GR24 for a LONG time so maybe the HoTT code is out of date but I don't think so.

manjoel, 2017-04-21:
I switched off and on again telemetry and added "set tlm_inversion=OFF" I will reset the MX-20 to bee sure not having an error here

My dump

# resources
resource BEEPER 1 C15
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource MOTOR 9 B10
resource MOTOR 10 B11
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 A09
resource SERVO 2 A10
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A03
resource PWM 2 A02
resource PWM 3 A09
resource PWM 4 A10
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1
servo 1 1000 2000 1500 100 -1
servo 2 1000 2000 1500 100 -1
servo 3 1000 2000 1500 100 -1
servo 4 1000 2000 1500 100 -1
servo 5 1000 2000 1500 100 -1
servo 6 1000 2000 1500 100 -1
[... 470 lines cut ...]
set level_limit = 55
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

hydra, 2017-04-22:
ok, thanks for the details, I will did mine out this weekend and give it a test to make sure it still works.   Please also try powering everything off and then on, with the HoTT firmware i was using the HoTT receivers only scan for telemetry for a short while after startup.

hydra, 2017-04-22:
problem confirmed, the issue is that the TX pin is not allocated due to the initial serial port mode.  This is a change from betaflight that broke it.  I will look at fixing it now.

hydra, 2017-04-22:
See also https://github.com/cleanflight/cleanflight/issues/2754

hydra, 2017-04-22:
actually, it turns out there are a number of issues:

1) TX pin not allocated correctly.
2) missing support for hardware bidirectional uarts
3) incorrect documentation in the spracingf4evo manual.
4) the only supported hardware uart ports are UART1/2/3.

to fix your issue, do not use UART5 on the SPRacingF4EVO and use uart1,2 or 3, connect both TX and TX pins and a diode as per the manual.  UART4 cannot be used in < CF 2.0.1

Cleanflight v2.0.2 adds:
* experimental support for HoTT on bidirection uninverted serial ports that support bi-directional transfers (i.e. using one wire, without a diode) - but not an always-inverted uart as found on the SPRacingF4EVO UART5.
* two-wire + diode all hardware uart ports (in addition to softserial/uart1/2/3)

The HoTT telemetry code needs updating, i created a seperate issue #2754 to track this.

hydra, 2017-04-22:
SPRacingF4EVO manual has been updated.

hydra, 2017-04-22:
Cleanflight v2.0.2 is out now - https://github.com/cleanflight/cleanflight/releases/tag/CLFL_v2.0.2

hydra, 2017-04-22:
I'll close this now, nothing more to do.  If it still doesn't work for you after reading the latest manual and using the latest cleanflight please create a new issue and reference this one.

hydra, 2017-04-22:
I should add, once you've tested it and got it working please leave a message here.

closed 2017-04-22 by hydra

manjoel, 2017-04-23:
I tryed little bit arround Uart 1 is used for pdb/osd.
So I tryed 3+4, but nothing worked. I resetted the board, I tryed with inversion.
Tomorrow I will look for an diode to verify, if there could be an other problem.