cleanflight/cleanflight #2746: Throttle Delivery to Motors by transmitter

rantonio90, 2017-04-20:
Hello,
I found what I think to be a bug after the lastest release to cleanflight 2.0 & 2.1
Configurations:
HEX+
SPRacingF3 Deluxe
Spektrum DX7 satellite on RX Serial 3
GPS M8N UBLOX on serial 2 at 57k baud
Serial Link with 433MHz module on serial 1 at 115k baud

After the update to the latest firwmare 2.x, arming the copter by switch starts the motors spinning at idle throttle but any throttle variation transmitted by the radio does not affect the motors, as if only the idle status is kept. With firmware 1.4.2 on the other hand everything was working properly. 

I checked the radio signals and ranges and they are the same as before updating.

The radio transmits correctly the signal as I can see the channels working on the receiver tab.

The motor ESC are correctly calibrated since I can change the motors' speed manually with the motors tab in the cleanflight configurator.

It looks like there is a check on something that prevents the radio signals to influence the motor updating routine, although I can see the model spinning when I move the sticks on the receiver tab.

hydra, 2017-04-20:
please provide a config dump.

hydra, 2017-04-20:
I just tested an SPRacingF3 with default configuration except for the RX type.  The oneshot motor signals are clearly visible and change when armed and sticks are moved, I don't think there's a code issue here.

Please use forums for support, if you still have an issue after verifying with other community experts please create a new issue and reference this one.  Perhaps also try comparing the new defaults with the settings you were using on 1.14.2 - much has changed!

closed 2017-04-20 by hydra