cleanflight/cleanflight #2747: Battery status incorrect. Battery level detection does not work on SPEV (QX70)

rbekkers, 2017-04-20:
Hi,

I noticed that my low battery detection did not work. The LED strip kept blinking pink. In the configurator the battery voltage is correct but the status is not connected. In CLI the status shows:
# status
System Uptime: 920 seconds
Voltage: 41 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500
SD card: Startup failed
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 1, config size: 1923, max available config: 4096
CPU:14%, cycle time: 132, GYRO rate: 7575, RX rate: 49, System rate: 9

[image]

As you can see the voltage is correct, but for some reason the battery is 0S and NOT PRESENT. 

Any idea what can be the cause?
Please let me know if I can help with additional information.

Thanks,
Rob

hydra, 2017-04-20:
yes. set `bat_detect_thresh` lower.  the default is 5.5v (`55`)  if you're on 1S/2S you may want to reduce this to about 2v (`20`)  From your detected voltage it looks like you're using 1S.

hydra, 2017-04-20:
Please use rcgroups for support in future.

closed 2017-04-20 by hydra

rbekkers, 2017-04-20:
You were so right. Thank you very much.