cleanflight/cleanflight #2752: Servotilt on SPRacing F3 acro not working

Hennyr, 2017-04-22:
I have configurate Servotilt and channel forwarding. (But also without channel forwarding same issue)
Servo does not react; in tab motors & servo i see servo 0+1 act on quadx movement.

I can not assign servo to a channel ; tried servo 0 > A02 ans Servo 1 > A03
Did it manual via resource.
When i ask resourse i get my reassign, but on dump NOT ?!? Also no change in tab Servos
Hereby the lists:
# dump
# Cleanflight / SPRACINGF3 2.0.1 Apr 19 2017 / 00:38:46 (229d76a)
name -
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00
mixer QUADX

mmix reset

servo 0 1000 2000 1500 100 -90
[... 548 lines cut ...]
resource MOTOR 6 B09
resource SERVO 1 A02
resource SERVO 2 A03
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00

What do i wrong.

Hennyr, 2017-04-22:
Update:
I have programmed the servo again; in dump no change, but resourse list shows:
# resource list
IO
A00: FREE 
A01: FREE 
A02: SERVO 1
A03: SERVO 2
A04: ADC_BATT 
A05: ADC_CURR 
A06: MOTOR 1
A07: MOTOR 2
A08: LED_STRIP 
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: MOTOR 3
A12: MOTOR 4
A13: FREE 
A14: SERIAL_TX 2
A15: SERIAL_RX 2
B00: FREE 
B01: FREE 
B02: ADC_RSSI 
B03: LED 1
B04: FREE 
B05: SERIAL_TX 11
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE 
B09: FREE 
B10: FREE 
B11: SERIAL_RX 3
B12: FLASH_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: FREE 
C15: BEEPER 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

Set channel_forwarding_start = 8  
Cann't forwading channel to servo's on my receiver. Also no activity on the controlls to the servo's in tab 'Motors & servo's'

Hennyr, 2017-04-22:
The Servotilt reacts after full reboot, but no action on my TX-controlls and cannot assign.
After save and leave the tab 'servo's', by return no assignment saved

hydra, 2017-04-22:
use the resource command to change some motor pins to servo pins.  documentation is out of date.

closed 2017-04-22 by hydra