cleanflight/cleanflight #2753: Throttle delivery to motors ISSUE

rantonio90, 2017-04-22:
Hello,
I have a problem with delivering the throttle to motors as you can see from the videos:

https://www.youtube.com/watch?v=KNWpODkTtTc

https://youtu.be/Wl0ah2QpvZE

What I see it is that changing the motor signal by using the motor tab in the configurator gives no problems.
However, the value does not work with the PID elboration.

Any idea?

hydra, 2017-04-22:
thanks for the detailed problem description and video, that really helps.

hydra, 2017-04-22:
from what i can see, it perhaps looks like your channel endpoints might not be configured correctly.

you are using these settings which are different from the defaults.
```
set min_check = 1050
set max_check = 1950
```

when your throttle channel is low, what is the value that appears on the receiver view for that channel?

hydra, 2017-04-22:
basically, you've set the min_check value too low, i was able to replicate this by having my RX throttle endpoint range go from 1100 to 1900 (on the TX) and then use your setting.  with the throttle low the value is never lower than the 'min_check' value of 1050 and so the pid controller never activates.

this is normal behaviour, your configuration is wrong.  ensure your channel ranges are correct i.e. 

leave the default values of min_check and max_check alone and configure your transmitter to send values between 1000 and 2000 for each channel OR use `rxrange`.

hydra, 2017-04-22:
Also @rantonio90 - and as previously metioned in #2746,  **_PLEASE use rcgroups/forums/etc for support._**

Wasting developer time like this is NOT appreciated. We have other more pressing issues to attend to - like coding!

closed 2017-04-22 by hydra

hydra, 2017-04-22:
Please also see these two images.
[image]

[image]

hydra, 2017-04-22:
@rantonio90 As a direct result if your behavior, I have now added an issue template to github which now appears when you attempt to create new issues.  I wanted something like this before but github did not offer the feature until recently.

At least something positive came out of this after all. 😄

rantonio90, 2017-04-22:
Is the min check a feature added with che 2.x version? Because same configuration was working with 1.42 and documentation is not that much detailed up to now
--
Sent from myMail for Android Saturday,

rantonio90, 2017-04-22:
BTW, it is not the problem of the rx range because my transmitters goes as low as 1000 with the throttle while the min was set to 1050.If I put the settings as you say I wouldn't even arm the copter because (as you say) the throttle is not low enough to allow arming. But the motors starts rotating when I move the sticker below the minimum.If I disable the turn on motors when arm, motors won't start rotating until I increase the throttle above the minimum (as it is correct).The problem is the PID that does not enable or the RC rate.I debugged the PID values and they never change when throttle increases after that motors spin.
I also try to disable the PID_at_min_throttle and nothing changes.
Are you using ST compiler? I'll try to check the code as well.
Antonio

hydra, 2017-04-23:
@rantonio90 no, it's a very old setting and has been around since the first release of cleanflight.

i think you're still mis-understanding the purpose of min_check and max_check.   they are used for all stick commands.  i.e. the system won't be able to detect if the throttle is low UNLESS the throttle value is LOWER than the min_check value.  the throttle must be considered LOW before the motors will do anything.

rantonio90, 2017-04-23:
Try this.

Use the default configuration and try to use the board.
It will work.

Then switch to HEX+ mix (I switched using the configurator)
The copter won't react to the receiver anymore.

Then switch to HEX X mix
The copter will start working again.

I don't know if it is a bug of the firmware or of the configurator but using default values this is what I get when using cleanflight 2.xx on the SPF3 

I reproduced the same problem changing few resources. It could be a problem related to resource usage.

You should try as well to verify  but I think it is problem in one of the define.

Also the HEX H does not work.