cleanflight/cleanflight #2755: Sonar not working op Sp Racinf F3 IO1

Hennyr, 2017-04-22:
I have activated Sonar, without Softserial, on version 2.0.1. Used port vor Sonar (io1 port 5&6)
No respons, measured with logic 4. Also no frequenty on trigger or echo.
do i have to use the special pads on the motherboard?
Herewith the dump:
# dump
# Cleanflight / SPRACINGF3 2.0.1 Apr 19 2017 / 00:38:46 (229d76a)
name -
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00
mixer QUADX

mmix reset

servo 0 1000 2000 1500 -50 -90
servo 1 1000 2000 1500 66 -90
servo 2 1000 2000 1500 100 -90
servo 3 1000 2000 1500 100 -90
[... 518 lines cut ...]
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

hydra, 2017-04-22:
yes it appears @borisbstyle broke it with this commit.  907a1840180de6840b5c5f11fafa57f3d6f944af

After it was commented out @mikeller later deleted it in ce6d0202e48b369b7af4d7a5c82b1fa1972451a3

It should never have been disabled in the first place.  I will look into this for CF v2.1.0

hydra, 2017-04-22:
Please test the binary here and report back.

https://github.com/cleanflight/cleanflight/pull/2757#issuecomment-296383944

Hennyr, 2017-04-25:
Hydra,
Tested version 2.0.3. 
Setting sonar, without softserial and telemetry: Sonar OK
Setting sonar, with softserial and without telemetry: Sonar OK
Setting sonar, with softserial and telemetry: Sonar OK

Have no telemetry on my transmitter yet. Have to investigate f i have done something wrong (IO1 pin 6 > Ibus sensor)

El-Tonno, 2017-05-03:
The changes in target.h kills BARO and MAG.
the if or line in line 97 is not needed -->
+#if defined(SPRACINGF3) || defined(FLIP32F3OSD)

El-Tonno, 2017-05-03:
I was NOT able to test SONAR positive with this changes.
Yes, the feature sonar is available but the sensor readings are faulty. Sonar reads roll or pitch changes in pulse waves instead of reading distance in a "sine wave" .
I´m using Sonar on CH7 / CH8 at IO2 with serial rx.

mehmetyldz87, 2017-05-05:
Also Sonar is not working with NAZE32 rev.5  with v2.0.2

hydra, 2017-05-05:
@El-Tonno has it ever worked for you?  What do you mean "pulse waves" vs "sine waves"?

El-Tonno, 2017-05-05:
Sonar worked fine on 1.4.2 
Now it seems to meassure something from the gyro. I can see digital pulse amplitude on the Sensors Tab if i flip the copter. It´s not reading the distance from the sonar module which would look like more than an analog sine wave amlitude when an objekt comes closer or the distance increases.

El-Tonno, 2017-05-05:
Hi just retested with modified target.h ... i removed line 97 so baro and mag works again.
The sonar problem is also SOLVED. For CF 1.4.2 the usb 5V is enough for correct sonar measurement.
CF 2.0.X needs also a connected lipo for sonar on my copter. Sorry my fault :-)
Sonar works on IO2 CH7/CH8 with CF 2.0.3 
... please just remove line 97 for the next release

closed 2017-05-12 by mikeller