cleanflight/cleanflight #2760: F3 OSD/PDB not working with F3 EVO

Windoofs, 2017-04-23:
The communication between the OSD and the FC seems not working.

OSD/PDB: SOF3 2.0.2
F3 EVO: SPEV 2.0.2
The OSD says "`Waiting for FC`" on the FPV monitor.
I followed the steps for OSD slave in the CF 2.0.2 release note.
When I uploaded the CF font, the OSD disconnects with this error: 
`Unrecoverable failure of serial connection, disconnecting...`

F3 EVO configuration: 
UART 1: MSP on with 115200 Baud
OSD activated

hydra, 2017-04-23:
reflash board boards with full chip erase.
configure evo: 
* enable msp on uart1 ports tab, save reboot
* enable osd on configuration tab, save reboot

works fine here.

Windoofs, 2017-04-23:
Thanks for your reply.
I repeated all steps like you mentioned above and in release note 2.0.2.
Same result. 
After the uploading of the CF front, the OSD/PDB disconnects itself, but without an error.

I triple checked the UART1 pins and the 5V/GND. Power is working and I can measure ~3,2V on the Tx and Rx pins. (I don’t have an oscilloscope at home)

hydra, 2017-06-04:
Did you ever get this resolved?  Certainly the code works on the pre-production and production boards I have here, guessing it's still a configuration issue or a hardware issue.  If you try the old CF 1.x firmware on both the FC and the OSD boards and that works then you can rule out the hardware  being an issue.  If CF 1.x doesn't work (i.e.  some of the OSD elements flash indicating no response from the FC) then it's probably a hardware issue in which case contact the retailer so the board can be checked/repaired/replaced.

Windoofs, 2017-06-04:
Thanks for your efforts.
I flashed v1.14.0 on the 3EVO and v1.14.0 on the OSD. After that the 3EVO works like normal but I am not able to connect the OSD via USB. `USB device not recognized.` Same with v1.14.2. 
Flashed both with full chip erase. 
I will contact SP Racing again.

hydra, 2017-06-06:
email received, once we get the board back we'll examine it and go from there, thanks for the update.

closed 2017-06-06 by hydra