cleanflight/cleanflight #2764: Mixer configurations issue

rantonio90, 2017-04-24:
# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

-- not all of the mixer configurations applied through the cleanflight configurator are working after the update to 2.xx firmware

# B - Steps To Repeat
BOARD SPRacing F3 Deluxe
-- update cleanflight to firmware 2.xx and configure your receiver settings
In my case I found the BUG by using spektrum receiver on serial 3 rx with taer1234 and 1024 us and I am arming the copter via the Aux1

1) Try the default configuration (QuadX) it should work and motors respond correctly to receiver.

2) Change the mixer configuration to HEX+ and try again. The copter will arm but the motors won't respond to commands anymore despite the throttle level they will always keep the signal to the min_throttle

3) Change again to HEX X. With this configuration the copter will work again and respond to motors.

4) Change to HEX H. This mixer does not work as well

# C - Video Links

https://youtu.be/KNWpODkTtTc

# D - Forum post links

-- provide links to forum posts if you have discussed this with others --

# E - Version Information

Cleanflight: v2.X.X
Configurator: v2.X.X

# E - Other information

-- SPRACINGF3 Deluxe
-- config dump + diff
[dump_diff_HEX+NOTworking.txt](https://github.com/cleanflight/cleanflight/files/951734/dump_diff_HEX.NOTworking.txt)
[dump_diff_HEXHNOTworking.txt](https://github.com/cleanflight/cleanflight/files/951735/dump_diff_HEXHNOTworking.txt)
[dump_diff_HEXX_working.txt](https://github.com/cleanflight/cleanflight/files/951736/dump_diff_HEXX_working.txt)
[dump_diff_Quad+working.txt](https://github.com/cleanflight/cleanflight/files/951737/dump_diff_Quad.working.txt)
[dump_diff_QuadXworking.txt](https://github.com/cleanflight/cleanflight/files/951738/dump_diff_QuadXworking.txt)
-- FUNFLY30A with BLHeli 14.9
-- Spektrum DX7 Satellite
-- Skektrum DX7
-- See SPRacing website: http://seriouslypro.com/spracingf3
-- windows 7 64bit

hydra, 2017-04-27:
what you might be seeing is the effects of `USE_UNCOMMON_MIXERS` being disabled for all targets at the moment.

the configurator needs updating.   in the meantime use the mixer commands in the cli (mixer CUSTOM, mmix, etc).

hydra, 2017-04-27:
e.g. for HEX 6X

source: https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L173-L179

```
static const motorMixer_t mixerHex6X[] = {
    { 1.0f, -0.5f,  0.866025f,  1.0f },     // REAR_R
    { 1.0f, -0.5f, -0.866025f,  1.0f },     // FRONT_R
    { 1.0f,  0.5f,  0.866025f, -1.0f },     // REAR_L
    { 1.0f,  0.5f, -0.866025f, -1.0f },     // FRONT_L
    { 1.0f, -1.0f,  0.0f,      -1.0f },     // RIGHT
    { 1.0f,  1.0f,  0.0f,       1.0f },     // LEFT
};
```

which easily translates into mixer commands as follows:

cli:

```
mmix 0  1.000 -0.500  0.866  1.000
mmix 1  1.000 -0.500 -0.866  1.000
mmix 2  1.000  0.500  0.866 -1.000
mmix 3  1.000  0.500 -0.866 -1.000
mmix 4  1.000 -1.000  0.000 -1.000
mmix 5  1.000  1.000  0.000  1.00
```

hydra, 2017-04-27:
i agree however, that the `mixer` command should not report all the mixers that are available when they are not compiled in.

the configurator should be updated to:
a) remove mixer setting from configuration page.
b) add a customer mixer tab which allows mixer settings to be configured.

we have talked about integrating this into the configurator UI for a while, but it's not been done as yet.

https://www.iforce2d.net/mixercalc/

hydra, 2017-04-27:
see also the docs:

https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md#custom-motor-mixing

mikeller, 2017-04-28:
> what you might be seeing is the effects of USE_UNCOMMON_MIXERS being disabled for all targets at the moment.

Good spot, @hydra. This was a puzzling one.

As a quick improvement, how about we change the mixer setting to be reset to default (QuadX), if a mixer preset for a not-compiled-in mixer is selected?

hydra, 2017-06-06:
this was done in 0b1383a7301a2011fa7afc438d3cac7c7529de92

closed 2017-06-06 by hydra