cleanflight/cleanflight #2767: Add support for disabling the UART1 port on the OSD.

hydra, 2017-04-29:
Use cause:
user has SPRACINGF3 ACRO/DELUXE
user wants to configure motors.
user powers OSD + FC + ESCS via battery.

what happens now is that the user is unable to configure the FC because the OSD is using UART1 connected to UART1 on the FC via stacking pins which means that UART1 on the FC cannot be used by the configurator.

workarounds:

a) on the FC add MSP to another serial port, use a USB -> serial adapter to connect to the FC via a different serial port.
b) put the OSD in bootloader mode, then configure FC as usual - keeps the bus-switch disabled.
c) erase the firmware on the OSD so that the bus switch is never enabled, put firmware back after configuring ESCs.

suggested implementations:
a) add button to UI so the next time the OSD powers on the bus switch is not enabled - a bit like spektrum bind currently does.
b) add buttons the UI to enable/disable bus-switch.

ratmole, 2018-01-15:
is there any update about this  enhancement?
Asking because i have a SPRACINGF3 + SP RacingF3 OSD/PDB and it's a little bit annoying to unstack the FC from the OSD...

mikeller, 2018-01-15:
@ratmole: Have you tried option c):

> c) put the OSD in bootloader mode, then configure FC as usual - keeps the bus-switch disabled.

Just pushing the boot button on the SPRACINGF3OSD board when powering up seems to be the least involved of the options listed.

ratmole, 2018-01-15:
FYI, there is no boot button soldered on my board.
great tip, i will add one on the board :)
thanks.

hydra, 2018-03-07:
closing due to lack of demand.  If you want this feature please add a 👍 to the original post and comment here.

closed 2018-03-07 by hydra