cleanflight/cleanflight #2773: Softserial and FrSky telemetry does'nt work anymore

m3us, 2017-05-01:
# A - Description

-- Softserial and FrSky telemetry does'nt work anymore on SPracingF3 Cf2.0.3

# B - Steps To Repeat

-- Turn on softserial, connect telemetry pin to soft serial port.Tried using telemetry_inversion in cli but didnt work.

The telemetry works on USART though. I am using it at the moment and thought id let you guys know about this issue.

hydra, 2017-05-02:
you're using the incorrect command, it's tlm_inversion now.  please pay attention to the get/set command output.

hydra, 2017-05-02:
also, you can use the slack group for support, see http://cleanflight.com/slack

closed 2017-05-02 by hydra

m3us, 2017-05-04:
I am aware of the change but tlm_inversion is on and yet there is no telemetry