cleanflight/cleanflight #2775: Tricopter Yaw Servo Not Responding in 2.0.x

bobthemac, 2017-05-02:
# 1. ARE YOU REQUESTING SUPPORT?

-- YES / NO --

# 2. If you answered YES to the above question go here instead:

https://www.rcgroups.com/forums/showthread.php?2249574-Cleanflight-firmware-for-STM32F3-based-FCBs-Check-First-Post-Please%21%21

Alternatively use other forums, slack or IRC for support.

# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

I have a tricopter and it worked before the update to 2.0.x I pluged it in to check some settings and got prompted to update so I did. Once updated the servo becomes unresponsive there is no signal sent to the servo. I am using a SPracingF3 Evo. Tested versions 2.0.0, 2.0.1, 2.0.2. It works fine in 1.14.2.

# B - Steps To Repeat

Upgrade to 2.0.x then apply yaw the tail servo will not move 

# C - Video Links

-- provide a link to a video showing your problem, as applicable --

# D - Forum post links

-- provide links to forum posts if you have discussed this with others --

# E - Version Information

Cleanflight: 2.0.X or commit SHA
Configurator: 2.0.3 or commit SHA

# E - Other information

OS: Debian Linux

hydra, 2017-05-03:
hi, your steps to repeat are not sufficient.  

how are you configuring the mixer?

can you provide a config dump.

it's likely you're running into a mixer configuration issue.  See also https://github.com/cleanflight/cleanflight/issues/2764

bobthemac, 2017-05-03:
I just set the model type to tricopter in the configuration tab.  Looking at the other issue if tricopters are seen as uncommon mixers that would explain it. I will have to take some time to get a dump as I reverted to 1.4.2 firmware as I know that works, I will have to re-flash and setup again.

lrodorigo, 2017-05-07:
Same problem here.
I also tried to remap SERVO 6 (default output for tricopter yaw) to MOTOR 4-8 but resource command seems to don't have any effect (_resource_ command show my new mapping but _dump_ command is still to **SERVO 6 NONE**).

lunebleue, 2017-05-08:
Hello, I have the same FC and using it with a tricopter I had to type this in CLI to put servo on output 4.

resource MOTOR 4 NONE
resource SERVO 1 A03
save

It took me a while as in motor tab in GUI I see it moving on servo 6 bar but in fact it is **S1** for yaw servo...

mikeller, 2017-05-10:
@lrodorigo: You have to remap servo 1 (and not servo 6) for a tricopter's yaw servo. (I know, tail servo is 'servo slot 6', but that gets mapped to servo 1, it _is_ confusing.)

Resources get initialised in numerical order, up to the first one that isn't assigned. So if you've got a resource mapped for servo 6 (but not 1 to 5), this will never get initialised.

DeetAus, 2017-05-14:
I am getting the same issue on an old naze32r5 board

The entire output mix has changed.

Previously the outputs (1thru 5 were) 
S1, S1, M1, M2, M3

With 2.0.2 the mix outputs are
M1, M2, M3   but no servo output

I tried the above suggestion:
resource MOTOR 4 NONE
resource SERVO 1 A03
save

but that doesn't appear to have changed anything
Looks like I will have to ground my tricopter until this can be resolved

victorivanovich, 2017-05-14:
Hello! Same thing happend to me. I have cc3d and I tryed configure as tricopter. Servo not moved and config where changed in servo panel not saved. Pleace check it.

mikeller, 2017-05-14:
You will have to remap your motors / servos to get the same setup as before 2.0.

For NAZE:

resource servo 1 a08
resource servo 2 a11
resource motor 1 b06
resource motor 2 b07
resource motor 3 b08

DeetAus, 2017-05-14:
I don't care if it isn't the same setup as before, just want the servo working, I have already re-done the ESC outputs to suit the new outputs

I'll try setting servo1 to b07 and see if that solves it

mikeller, 2017-05-14:
@DeetAus: Sorry, I was not entirely clear in my last post. It will **not** work unless you go back to the wiring that worked for < 2.0. Motors on outputs 1 / 2 / 3 and Servo on output 4 is **not** supported.

(The reason for this is a hardware limitation: On the NAZE board, outputs 1 and 2 share the same timer, and outputs 3 - 5 share another timer. Motors and servos cannot share the same timer. As a consequence, there is no way to use 3 motors and servos, and have the motors use outputs 1 or 2.)

DeetAus, 2017-05-14:
Thanks, will test and get back to you

DeetAus, 2017-05-14:
OK
Thats got the servo and motors back to where they used to be, and the servo is no working BUT

Its correcting in the wrong direction and I cannot get it to reverse via the servo tab of the configurator

mikeller, 2017-05-15:
Good to hear it's working. And yes, tricopter support in Betaflight is somewhat clumsy, there is no equivalent for the `smix reverse` CLI command in the Servos tab.

Strommy, 2017-05-20:
I am having the same problem. Using latest CF and latest update. Aircraft set to tri copter. Connected the correct way and had to enter CLI as below to set servo and motors to correct channel:-
resource motor 1 none
resource motor 2 none
resource servo 1 a8
resource servo 2 a11
resource motor 1 b6
resource motor 2 b7
resource motor 3 b8
resource motor 4 b9
save

All motors spin up correctly now and servo is listed as servo 6 in motors page. Servo 6 graph  moves correctly when yaw is activated on screen but no reaction from servo. Running digital servo and also tried with a analog one to no avail. Naze32 rev6 acro. (Tried a second one but still same issue. Any help would be great. Even if someone has a working CLI code dump so I can load to mine.
Thankyou in advance

mikeller, 2017-05-20:
Have you set `tri_unarmed_servo = on`?

Strommy, 2017-05-20:
I tried that but it is an unknown command in CLI.
below is dump of CLI.
# dump
# Cleanflight / NAZE 2.0.2 Apr 22 2017 / 14:41:48 (4328b13)
name -
resource BEEPER 1 A12
resource MOTOR 1 A08
resource MOTOR 2 A11
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 A02
resource PWM 4 A03
resource PWM 5 A06
resource PWM 6 A07
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A06
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
mixer TRI

mmix reset

servo 0 1000 2000 1500 100 -1
servo 1 1000 2000 1500 100 -1
servo 2 1000 2000 1500 100 -1
servo 3 1000 2000 1500 100 -1
servo 4 1000 2000 1500 100 -1
servo 5 1000 2000 1500 100 -1
[... 370 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

any help would be great.
thankyou for helping **mikeller**

mikeller, 2017-05-21:
Looks like you did not save your resource remappings:

>resource MOTOR 1 A08
resource MOTOR 2 A11
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE

(Also, `tri_unarmed_servo` is already on in your config.)

Strommy, 2017-05-21:
Thats weird as i just checked resource and it states it as. after i changed settings i always save. but when i do a "dump" it comes up as posted above...

# resource
resource BEEPER 1 A12
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource SERVO 1 A08
resource SERVO 2 A11

Just to confirm power going through to the servo can go through unused channels (ie out put 2 or 6) ?? as i have tried putting power through receiver to power naze and also tried through unused channels. (just incase the bus is similar to the kk board where as 2-6 are not on same bus as output1) 
i have also tried the "triflight_0.7_Beta1_NAZE.hex" that rcexplorer recommends but still to no avail.
I have been looking for a stevis hex for naze. I just cant understand why after the hex has been loaded and the model set that you have to remap outputs.

mikeller, 2017-05-21:
Try saving it again. Also, note that if you reset the config, the resource mappings will be reset as well.

> I just cant understand why after the hex has been loaded and the model set that you have to remap outputs.

That's because 2.0 has removed the custom resource mappings for non-quad models that existed for a dozen or so boards, and introduced user configurable resource mappings.

At a later stage there will be support for preset resource mappings for models like tricopters. But with over 100 boards now supported, and all of them requiring their own custom mapping, the effort for this (including testing) will be quite large.

Strommy, 2017-05-22:
okI have tried now tried using Cleanflight 1.10 to 1.14 and older versions of releases to no avail. i have also tried it on betaflight 3.0 with resource remapping on all but no movement from servo. Servo works perfectly on tester and so does analogue servo. so no faults there.. tried to also load the dump from rcexplorer settings but no reaction. ensured i saved it and board rebooted and saved as it usually does with my quads. tried power on the channel 2 output and channel 6 (not at same time) and also by powering it through receiver ppm input. channel 6 on motor page of both cleanflight and beta flight register the movement through yaw channel but no out put is produced. enabled channel A8 for the servos. 
starting  to Just wonder if the NAZE32 rev6 board works with a tricopter? if not what suggestions are there except the kk which i use already.

mikeller, 2017-05-22:
> starting to Just wonder if the NAZE32 rev6 board works with a tricopter?

Yes they do, I am using one.

Different question: Have you tried moving the servo by changing the mid point of slot 6 in the 'Servos' tab?

Strommy, 2017-05-22:
yes. enabled expert mode. went to servo tab and adjusted servo 5 to 1800 and it changed the mid point on the motors page of servo 6 (which is my yaw channel as it moves the bar on movement of yaw. I thought servo on tri was servo 0 on servos tab.)
So i reflashed the naze, loaded current cleanflight configurator and current version 2.0.2,  changed the resourses to allocate motors and servos to which the motors work. servos tested using servo tested and even tried to power servo independantly and just use signal wire to tell servo position. no avail.
i have tried to enable servo tilt to see if that works but nothing. and just tried my other 2 boards but same thing.
using ppm input with spectrum dx6 reciever. servo is a bluebird digital 385dmax or 380 max for analogue.
when i go to cli screen and command dump it states the dump as previously posted but if i check resourse everything is correct.
if i go to the servo tab and select a checkbox and save, it will not be there next time i enter the tab but the midpoint settings stay.
you are using a tri. which cleanflight configurator and version of flash are you using? board is an acro too.

rejo54, 2017-05-30:
Strommy , 
I had the same problem. The motors will spin up, the server control worked in Cleanflight and was listed as servo 6, but no servo movement on my tricopter. Thanks to this forum I got it working now :).  Nothing was saving when I made changes in cleanflight.

I opened Betaflight CLI  I did the following. 

resource MOTOR 4 NONE
resource SERVO 6 A03

A03 was the resource for motor 4.

On the servo tab I set servo 6 to ch 4 (ticked the box)
changed the values to -45/45 from -90/90

Hope this helps.

Strommy, 2017-05-30:
I have tried everything possible. It all saves. And the resource listed in cli is correct. All up it just doesn't want to play might be the hk boards or something... so I bit the bullet and put the kk back in and updated to 811 software to get some of the same features of horizon mode (slmix!)
Will keep using the naze for my quads but as for tricopters I will try another day when there is more info and more data dumps on the Web so I can copy it into my cli and test that. But happy with the kk as it works first time everytime. Just wish they were 32 bit too.
Thankyou all for helping out.

rejo54, 2017-05-30:
@mikeller: Thank you for your input. I never stated that the resources had to be the same as mine, my point was servo 6 never had resources in the CLI and I cleared the resources from motor 4 and used those for servo 6 and it worked. 
As Stormmy had posted above:
"ll motors spin up correctly now and servo is listed as servo 6 in motors page. Servo 6 graph moves correctly when yaw is activated on screen but no reaction from servo."
"resource SERVO 6 NONE"

It seems like the same problem I had. Just because you could not help him does not mean others input may not!

mikeller, 2017-05-30:
@rejo54: I am a bit surprised by your statement that you when using servo 6. For the tricopter mixer, it is servo 1 that is used. There is some confusion between 'servo slots' as listed in the configurator, and servos, which is how resource management sees them. For a (somewhat lacking) explanation have a look at https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md#custom-servo-mixing 

(Also, the resource for servo 6 will not be initialised unless you define resources for servos 1 - 5 too.)

hydra, 2017-06-06:
afaikt there's no issue here other than a configuration issue.  yes we know the configurator could be more helpful in this regard and can be addressed in the future when we have time.  If this isn't the case and there's a genuine bug here then please create a new issue with exact steps to repeat and reference this one in the description.

closed 2017-06-06 by hydra

SteCowell, 2018-02-02:
I've been trying to setup a Quantum Illuminati 32 (essentially a NAZE 32) on a trictopter for a number of weeks without success. The tricopter motors worked fine but the YAW servo just would not cooperate. I've read numerous forums and papers but it was thanks to the comment by mikeller (May 14, 2017); which explained the timer issue and provided the mapping, that my setup has been successful. 

Illuminati 32 connections: 
- pin1 to servo
- pin3 to motor 1
- pin4 to motor 2
- pin5 to motor 3

Resource mapping (as per mikeller comments):
- resource servo 1 a08
- resource servo 2 a11 
- resource motor 1 b06
- resource motor 2 b07
- resource motor 3 b08

Many thanks mikeller and to all other contributers.