cleanflight/cleanflight #2786: SPRacing F3 OSD/PDB + F3 EVO    current sensor value issue

lunebleue, 2017-05-07:
The current value displayed by the OSD when no motor is running is 19A !!!

When I connect the USB to OSD/PDB board and go to cleanflight GUI (Power & Battery tab) the value of current is CORRECT with 0.6A

When I connect the USB to F3 EVO ang go to cleanflight GUI (Power & Battery tab) the current display is WRONG 19A 

It is this wrong one that is diplayed in my googles and that is used for mAh calculation :(

No problem with voltage display

Both board that I ordered recently directly to SPRacing have both the latest firmware on it.

Thanks for your help on this.
[image]
[image]

tonipalmgren, 2017-05-08:
I have the same configuration, but cannot see proper current value anywhere. @hydra suggested to define the current meter scale by adding:
#define CURRENT_METER_SCALE_DEFAULT nnn
in src\main\target\spracingf3osd\target.c, and rebuild the firmware. I measured the resistor next to the current meter IC, and got 42,7 kOhms, which means the "nnn" should be 206. Added that to the target.c file, but the firmware wouldn't compile. Added it to the target.h file and it compiled, but it didn't have any impact. I still get a bit over 16 Amps when the motors aren't spinning. Can try rebuilding and flashing, if anyone has any ideas.

hydra, 2017-05-29:
This is a configurator issue, the configurator needs updating to support changing of the sensor offsets and scales, once this is done you'll be able to configure it correctly.

closed 2017-05-29 by hydra

yo8aiv, 2017-06-10:
hello, 
Has this issue been resolved !!!    If yes how ???.
im in the same situation 
- current tab is disabled 
- Amps show arround 19 - 20.12A  .. oscillating., no effect on starting motors.
Cleanflight / SPRACINGF3EVO 2.0.2 Apr 22 2017 / 14:55:51 (4328b13).

Thanks.

ecelkis, 2017-06-22:
I'm Having the same issue.  SP Racing F3EVO with SP Racing OSD/PDB.  It seems as if the current and voltage readings are coming from the FC rather then the OSD board.  That means that the current sense resistor needs to be connected to the F3EVO to work.   This can be done as a workaround but not the cleanest patch.  I have all the most recent stable firmware installed on both boards as well as the latest configurator.  For some reason when selecting the "use onboard ADC" option in the OSD board setup, the current and voltage from the FC are still being used for the OSD display.  Please Help.  Thanks much.  Love your work!

ecelkis, 2017-06-22:
Should I open a new issue on this?

hydra, 2017-06-23:
@ecelkis that is not correct.  there is an option to use 'OSD SLAVE' as the source in the configurator.

ecelkis, 2017-06-24:
Dominic,  Yes that is correct however the F3EVO running 2.0.3 is not using api version 1.36 so no 'OSD SLAVE' in the drop down.  Any suggestions?  Thanks again!

ecelkis, 2017-06-26:
So I figured it out.  You need to build and flash the code from the osd-slave-msp-current-meter branch to your FC.  This code has the correct API version.  I tried a test flight this weekend and everything seems to work fine.  I'm assuming at some point that branch will be merged with the master branch...

richard-scott, 2017-10-19:
@tonipalmgren you said:

> I measured the resistor next to the current meter IC, and got 42,7 kOhms, which means the "nnn" should be 206.

What formula do you use to work that out?

tonipalmgren, 2017-10-19:
@richard-scott 

I don't know the formula. When this was an issue, Dominic pasted the common values on Slack so I could set the correct one into the config before compiling the firmware.

currentMeterScale = 366;  // INA139 - 73.2k / 90A //
currentMeterScale = 275;  // INA139 - 55.0k / 120A //
currentMeterScale = 206;  // INA139 - 41.25k / 160A

I'm sure there is a formula, but that 206 was close enough for me.

richard-scott, 2017-10-20:
@tonipalmgren 

Ok, thanks for taking the time to reply.