cleanflight/cleanflight #2787: PID adjust in flight changes to much values each movement

McGiverGim, 2017-05-08:
# A - Description

The RC Adjustements in flight can't be used, at least for some kind of switch. I have a FlySky FS-i6S TX, and I use the VrA and VrB wheels to change the P and D values.
In CF 1.x I had no problem with this, each movement and release make one or two beep, and the value change 1 unity.
Now, in CF 2.x each movement and release make only one or two beep, like before, but makes several changes, this is a BlackBox image of a single movement and release:
[image]
I don't know if the result will be the same for a 3 pos swich.

# B - Steps To Repeat

Configure a Vr or similar (with behaviour like a stick) to modify some of the PID values. Use it. 

# E - Version Information

Cleanflight: v2.0.2

# E - Other information

diff:
```
# diff
# Cleanflight / NAZE 2.1.0 May  4 2017 / 18:20:08 (a736f87)
name Robocat
feature -RX_PPM
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
serial 0 1 19200 57600 0 115200
serial 1 4160 115200 57600 115200 115200
serial 30 2 115200 19200 0 115200
aux 0 0 0 1950 2050
aux 1 28 1 1750 1850
aux 2 1 1 1050 1450
aux 3 1 1 1550 1650
aux 4 2 1 1450 1550
aux 5 6 1 1150 1250
aux 6 6 1 1550 1650
aux 7 3 1 1050 1450
aux 8 3 1 1550 1650
aux 9 5 1 1550 1650
aux 10 5 1 1050 1350
aux 11 10 1 1250 1350
aux 12 11 1 1050 1350
aux 13 13 1 1850 1950
aux 14 13 1 1250 1350
aux 15 26 3 1950 2050
adjrange 0 0 1 1650 1850 7 5
adjrange 1 0 1 1650 1850 6 4
rxfail 5 s 1200
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 150
set mag_hardware = AUTO
set magzero_x = -36
set magzero_y = 43
set magzero_z = 296
set baro_hardware = AUTO
set min_check = 1050
set max_check = 1950
set fpv_mix_degrees = 15
set serialrx_provider = IBUS
set blackbox_rate_denom = 2
set min_throttle = 1045
set max_throttle = 1980
set failsafe_throttle = 1350
set failsafe_procedure = AUTO-LAND
set current_meter = NONE
set disarm_kill_switch = OFF
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set deadband = 3
set debug_mode = NOTCH
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 206
set anti_gravity_gain = 2
set setpoint_relax_ratio = 95
set dterm_setpoint_weight = 80
set p_pitch = 62
set i_pitch = 55
set d_pitch = 38
set p_roll = 50
set i_roll = 45
set d_roll = 33
set p_yaw = 90
set i_yaw = 50
rateprofile 0

set rc_expo = 25
set thr_expo = 55
set roll_srate = 55
set pitch_srate = 55
set yaw_srate = 50
set tpa_rate = 0
```

hydra, 2017-10-21:
@McGiverGim is this still broken in master? can you check again please?

McGiverGim, 2017-10-21:
@hydra I don't have the NAZE that give me this problem. I can assure that with the omnibusf4sd this does not happen, so this can be closed.

McGiverGim, 2017-10-23:
I close the issue because I don't have the material to reproduce it again.

closed 2017-10-23 by McGiverGim