cleanflight/cleanflight #2789: Receiver don't work

luismi7777, 2017-05-08:
# 1. ARE YOU REQUESTING SUPPORT?

-- YES / NO --

# 2. If you answered YES to the above question go here instead:

https://www.rcgroups.com/forums/showthread.php?2249574-Cleanflight-firmware-for-STM32F3-based-FCBs-Check-First-Post-Please%21%21

Alternatively use other forums, slack or IRC for support.

# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - I have a radiolink rd6sm connect via sbus to a naze 32 rev6 dof10

And don't appears in receiver tab all bars are stopped

# B - Steps To Repeat

Port tab are configured uart 2 serial rx
Setup tab I select  spectrum receiver -》sbus

And in receiver tab  the bars are stopped and the  throttle  is 885
In my radio (at9) are linked and I see the status and  is correct 

I reflasheD and link  with the radio a lot of times and nothing 

# E - Version Information

Cleanflight: v2.02 
l naze 32 rev 6 dof10

-- dys sn20
-- radiolink  r6dsm
-- at9
Windows 7 ,mac lion , Mac captain

closed 2017-05-29 by hydra

lexusburn, 2017-11-17:
Any solutions regarding this? Have the same problem with R6DSM and Radiolink AT10II and Naze32Rev6Dof10 and FuriousFPV Nuke. Both won't work with and without SBUS inversion.

ledvinap, 2017-11-18:
My radiolink receiver sbus implementation is broken, flags bits are swapped.