cleanflight/cleanflight #2790: Micro Scisky FC not connecting

ghostface, 2017-05-09:
Same problem as with betaflight (since you are based on them now): 

Flashing works fine but the configurator does not connect.
I revertet back to <2.0.0 now in order to be able to use my FC :(

> Tested on archlinux and windows in chromium and chrome.
> Cleanflight flashes and connects fine so does 2.9.1 of betaflight (using Alienflight F1 target).

https://github.com/betaflight/betaflight/issues/2811

Any help appreciated!

krobinson384, 2017-05-19:
I am having the same issue with my Micro Scisky, it worked before the app update!

mikeller, 2017-05-20:
@krobinson384: See discussion re using NAZE target for this board in https://github.com/betaflight/betaflight/issues/1061.

Duplicate of https://github.com/betaflight/betaflight/issues/1061 .

closed 2017-05-20 by mikeller