cleanflight/cleanflight #2791: cleanflight  can not make by gcc 4.9.3

xdwgood, 2017-05-10:
my gcc tool version is:4.9.3
Is mean I can not make ? only gcc 6.3.1 can make it?
Do you have good idear to resolve the issue?
make/tools.mk:296: *** **ERROR** your arm-none-eabi-gcc is '4.9.3', but '6.3.1' is expected. Override with 'GCC_REQUIRED_VERSION' in make/local.mk or run 'make arm_sdk_install' to install the right version automatically in the tools folder of this repo.  Stop.

ledvinap, 2017-05-10:
You can comment out the line make/tools.mk:296 to suppress this error.  But 4.9.3 is quite obsolete, you may encounter unexpected problems. Consider upgrading your toolchain (or use make arm_sdk_install to install local copy of supported version)

xdwgood, 2017-05-10:
@mikeller @ledvinap Thank you very much for your reply

closed 2017-05-10 by xdwgood

xdwgood, 2017-05-10:
@mikeller I am in China, how do I buy SPRACINGF3 or a whole set of UAVs?

xdwgood, 2017-05-10:
when I try to make CC3D or NAZE, I can successfully get hex file.   
But when I  make  SPRACINGF3, I can not get hex file.
make[1]: *** [obj/main/cleanflight_SPRACINGF3.elf] Error 1
make[1]: Leaving directory `/home/nuaa/cleanflight/cleanflight-master'
make: *** [hex] Error 2

xdwgood, 2017-05-10:
I commented on makefile #275, and then I was able to get the hex file successfully.