cleanflight/cleanflight #2792: SP Racing F4 Evo isn't moving any servos as fixed wing

sd1cko, 2017-05-10:
# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

SP Racing F4 Evo isn't moving any servos as fixed wing

# B - Steps To Repeat

Vcc pin has been bridged to output power to all 8 pwm output pins. Configured and set up FC in Cleanflight as per instructions and selected Flying Wing configuration. Check receiver inputs etc are correct as per instructions, setup arm switch and other modes, and checked output on the motors screen which show to be working, with no physical movement to any servos. I've tried with several known working servos too, and also checked for 5v output on the pwm pins and this is working.

# C - Video Links

Not really relevant.

# D - Forum post links

# E - Version Information

Running cleanflight 2.0.3 and firmware v2.0.2 SPRACINGF4EVO

# E - Other information

As stated, flight controller is SP Racing F4 EVO

Config dump:

# resources
resource BEEPER 1 C15
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource MOTOR 9 B10
resource MOTOR 10 B11
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 A09
resource SERVO 2 A10
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A03
resource PWM 2 A02
resource PWM 3 A09
resource PWM 4 A10
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
[... 570 lines cut ...]
set level_limit = 55
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

mikeller, 2017-05-10:
From the look of it, your configuration does not have the servo outputs mapped to MCU pins that are actually connected to pins / pads on the board (or at least I can't find them mentioned in the documentation). Your best shot (taking potential timer conflicts into account) is to connect the servos to PWM 5 and 6 of the 8 ESC / PWM outputs, and remap the servo pins accordingly with `resource servo 1 b06` and `resource servo 2 b07`.

sd1cko, 2017-05-10:
Thank you for your (very fast) response Mikeller :)

I will try this and get back to you. Is this just an over sight in the firmware for the SPF4EVO? 

I did look to see if I could find which MCU pins were mapped to which pwm outputs but I couldn't find any information on this. Am I to assume that the motor will use the correct resource? At the minute I have this completely on the bench with my laptop for testing / set up, however my motor is mounted to my wing so the test for the esc / motor output is yet to be done (one thing / problem at a time). I would assume this would be fine though or else even quads wouldn't be able to fly if motors were outputting to the wrong MCU pin.

Thank you again Mikeller, I shall report back.

sd1cko, 2017-05-10:
I can confirm this does indeed work! Thank you ever so much Mikeller.

Is there any chance that you could perhaps share the MCU pinout to PWM pins in case I need to remap anything else in the future such as motors or other servos etc?

Thank you again, your help has been massively appreciated.

mikeller, 2017-05-10:
You're welcome. :-)

I don't have much experience with fixed wing aircraft, but I am assuming that the mixer for flying wing will be outputting the signal for motor 1 onto the ESC / PWM 1 pin.

It's not so much an oversight as it is a rationalisation that had to be made when user definable resource mappings were implemented. Before this, the dozen or so flight controller models that supported mixers that included servos all had the pin assignments for servos hardcoded in the firmware, since they had to be all different, to work around the fact that on most boards ESC / PWM pins share timers, and timers cannot be shared between motor and servo pins. With more than 100 boards that are supported now, maintaining these hardcoded assignments in the firmware just isn't an option any more, so for the moment it is up to the user to assign motor pins to servos if required.

In the future we will probably be looking into defining 'presets' for non-quad mixers in the configurator, and using these presets to configure board-specific pin mappings.

Pin mappings can all be looked up in the target definition for the board: https://github.com/cleanflight/cleanflight/blob/master/src/main/target/SPRACINGF4EVO/target.c (it's open source, after all. ;-)

sd1cko, 2017-05-10:
> SPRACINGF4EVO/target.c (it's open source, after all. ;-)

Excellent, thank you Michael!

This is all fairly new to me, I've not had anything RC since I was a
teenager, and I'm now getting my son into scratch building, and wanted to
use a FC to get him into the air gently using angles and horizon.

I looked through documentation and how to's etc and nowhere did it mention
having to manually assign outputs (unless I missed it?). Even youtube
videos I have watched just show the servos magically working after
selecting the Flying Wing configuration (albeit on other boards) :)

I have had experience with programming pic chips etc from other hobbies so
I have a basic knowledge of some of the requirements which is helping, but
I definitely still have a lot to learn. Documentation seems very scattered
/ scarce for fixed wings, and I think I made it worse for myself by getting
a relatively new FC, thus making the documentation even more scarce. We
live and learn. Hopefully others with similar problems may find this and it
will help them. I did find similar threads for the Naze board where the OP
found no resolution due to nearly everyone use FC's for quads so could only
offer pointers.

Thank you again Michael. I really really appreciate you taking the time to
help me.

Mob: 07803 557021
Tel: 01757 638961
email: steve@syntaxrepairs.co.uk

hydra, 2017-05-29:
thanks @mikeller

closed 2017-05-29 by hydra