cleanflight/cleanflight #2794: UBLOX GPS Failure after upgrade (softserial)

apavel3458, 2017-05-10:
# A - GPS stopped working during an update to Cleanflight 2.0.2 (from Cleanflight 1.x).  GPS is powered (light indicators are working) and likely has GPS lock, but cleanflight won't recognize it.  Cleanflight also won't save UBLOX setting, and CLI commands related to GPS are now gone.

# B -

1. My receiver is PPM mode, so I connected channel 7 TX to RX of GPS, and channel 8 RX to TX of GPS. 
2. Connected 5V power supply to GPS. 
3. Enabled SOFTSERIAL
4. clicked "GPS" in PORTS window.
5. Enabled GPS in configuration, and changed it to UBLOX, AUTO BAUD, AUTO CONFIGURE. 
- Tried various baud rates (even low ones, given that this is soft-serial), no help. 
(see screenshots)

Cleanflight does not SAVE my UBLOX setting.  When I disconnect and reconnect it goes back to NMEA. 

I followed the documentation and the command for set gps_provider=UBLOX no longer works (variable unknown). https://github.com/cleanflight/clean...er/docs/Gps.md, other variables are missing too. 

# C - Video Links

[image]
[image]
[image]

# D - Forum post links

Posted this on the forum, but no one knows.
https://www.rcgroups.com/forums/showthread.php?2249574-Cleanflight-firmware-for-STM32F3-based-FCBs-Check-First-Post-Please%21%21/page1056#post37488456

# E - Version Information

Cleanflight: v2.0.2
Configurator: v2.0.3
Naze 32 (Skyline32 edition)
GPS is a UBLOX Neo6 series 

# E - Other information

CLI Dump:
[cli.txt](https://github.com/cleanflight/cleanflight/files/991398/cli.txt)

mikeller, 2017-05-10:
Due to the very limited flash space available on NAZE and other F1 based targets, GPS support has had to be removed from these targets in order to make them fit into flash. You'd have to build your own version of the firmware for this target with GPS enabled (and other features disabled to make it fit). Or else, upgrade your board to a more current one with an F4 MCU, to get back support for GPS and other advanced features.

apavel3458, 2017-05-10:
Hi there, thank you for your reply.  this is helpful. 

1.  Can you point me in the direction of installing the old cleanflight 1.x  (because the configurator no longer recognizes the old cleanflight software).  

2. If GPS feature was disabled, why do you still have it in the configurator?  I spent about 4-5 hours trying to get it to work last night, soldering different connectors, checking everything over and over.  

3.  Can we update the documentation to reflect this?  I didn't see it mentioned in the update logs (I may have missed it).  It would be nice for GPS-requiring users to know that it was disabled. 

Thanks!
Pavel

mikeller, 2017-05-10:
1. I suspect that GUI support is no longer available for cleanflight 1.x. You can still use configurator to install the old firmware, and then use the CLI to configure it.

2. This is a shortcoming of the current firmware / configurator implementation. There is a new protocol for firmware - configurator communication in the works that will improve this. (Currently, there is no way for the configurator to query the firmware for a list of the features that are supported.)

3. Yes we can. Feel free to submit a pull request to update https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20Naze32.md . Unfortunately documentation is often lagging behind the reality, since people tend to allocate more of their limited time to writing code than to updating the documentation.

apavel3458, 2017-05-10:
Darn... mikeller thanks for your reply.

I may try Baseflight again. 

After a year of using Cleanflight and loving it...its a shame to see GPS be disabled. 

It is quite important to me because I want RTH feature in case I lose signal.  Heaven forbid my bird falls and harms something or someone because I lost signal and GPS is no longer supported. 

It's also too bad older 1.x stable firmware was actively deprecated (no longer has any usable UI).  I think it may have been a bit premature to disable support for version 1, given that everyone loved it (including myself), and that 2.x has quite a few bugs to work out still. 

Thanks again for clarifying everything.

mikeller, 2017-05-11:
Well, the reality is that giant steps were made with respect to flight performance, most existing features were improved, and new and exciting features were added. As a consequence of this, the F1 MCU that was perfectly capable of running the old code is not adequate to run the improved code that is in 2.0.
It's a bit like trying to run Windows 10 on that PC that you bought 10 years ago, and that ran so well with Windows XP way back when. (Also, like Microsoft, people here are choosing to allocate their time to improving the new version, at the expense of supporting deprecated versions.)

Before you discard 2.0 entirely, I suggest you take your craft out to the field, and have a quick fly with it. Chances are that you will be blown away by the improvements in flight performance, and will see why it makes sense to upgrade your board to one that can deliver this performance without having to cut corners.

On a side note, since you seem to be interested in advanced GPS features like RTH, you should have a look into the iNav project that focuses on advanced navigation: https://github.com/iNavFlight/inav

apavel3458, 2017-05-11:
Thanks Mike,
iNav is a cool project looks like. They are also talking about stripping down F1 targets as well. 

Can you recommend a good flight control board that I can upgrade to? . I've been using naze32 (skyline32 edition).

https://m.banggood.com/EMAX-Skyline32-Naze32-Advanced-Flight-Controller-Barometer-Support-Cleanflight-Baseflight-p-981145.html?rmmds=search

You seem to be quite experienced, and I don't know much about them.

Thanks!!
PA

mikeller, 2017-05-11:
If you are interested in using advanced GPS features, it is a good starting point to look for a board that is supported by iNav, since iNav GPS support is much more advanced than cleanflight / Betaflight: https://github.com/iNavFlight/inav/wiki/Supported-boards (Also, iNav supports less boards than cleanflight, so this limits the potential choices quite a bit.)

Out of these I would then choose a board that is based on F4 and supported by cleanflight, so that, if you decide you want to stay with cleanflight, you get a board that has got enough flash space to allow for all possible advanced features (including GPS) to be included in the firmware.

(Out of the boards that meet these criteria, I am personally flying BLUEJAYF4 (with Betaflight) and BEEROTORF4 (with Betaflight / iNav), and happy with both of them.)

apavel3458, 2017-05-11:
Thanks Mike, 

That was super helpful!  I'll look into getting one.  Ill try iNav for now since they have some F1 support still.

All the best to you!

Pavel

mikeller, 2017-05-11:
Have fun!

F1 support is kept around (in a very limited way) just for cases like yours, so that users can try never versions before committing to spending money and time on getting and installing a new board.

fftunes, 2017-05-11:
@apavel3458 "It's also too bad older 1.x stable firmware was actively deprecated (no longer has any usable UI). I think it may have been a bit premature to disable support for version 1, given that everyone loved it (including myself), and that 2.x has quite a few bugs to work out still."

You could still manually install an older version of the configurator to use an older version of cleanflight.
But yeah, you're absolutely right - "premature" is a mild way to put it. Annoys the heck out of me having to answer the same question over and over in every forum i visit...

McGiverGim, 2017-05-11:
Here you have the configurator for 1.x: https://github.com/cleanflight/cleanflight-configurator/tree/master_v1.x

tonipalmgren, 2017-05-11:
Slightly off-topic, but is there a place where we could create a basic how-to on installing the older version? This is the most frequently asked question in the Cleanflight Slack. I'd much rather paste a link telling people where to go than every time write down the same instructions. Same goes for upgrading the firmware.

fftunes, 2017-05-11:
@tonipalmgren There's youtube videos about this... i know painless360 has one with the necessary links in the description.

tonipalmgren, 2017-05-12:
@fftunes Of course, but videos are the least effective way of transferring information. Usually in a 10-minute video there's around a minute or two worth of information you actually need, and this information can easily be told with a few lines of text and a couple of screenshots. Videos are like making a film of the 200 vacation photos you took and then endlessly trying to seek to the few good ones you really want see. :) And you can't easily update the video with new information (when you get the other 200 pics your travel buddy took).

I've watched a lot of painless' videos and they are nice, but, as with every video, the information you're after could often be compressed into a few lines of text and a couple of images, and this is where written documentation will always be superior.

Sollex-21412, 2017-05-19:
How can I build FW for Naze32 with enabled GPS. What should I do? Is there any instructions?

mikeller, 2017-05-19:
@Sollex-21412: Yes there are: https://github.com/cleanflight/cleanflight/blob/master/docs/Customized%20Version.md

closed 2017-05-29 by hydra