cleanflight/cleanflight #2796: Firmware 2.0.2 not worked on CJMCU

ErfanDL, 2017-05-12:
Hi. please fix firmware 2.0.2 on CJMCU. when uploading firmware motors run selfi

mikeller, 2017-05-12:
Is that 'during the upload'? Please read the safety instructions, and never have a battery connected _and_ props on when flashing firmware.

ErfanDL, 2017-05-12:
@mikeller
No. upload is successful but when I turn on cjmcu motors start by itself

mikeller, 2017-05-12:
Ok. Looks a lot like the brushed motors are not detected properly. Can you connect it to the configurator and check what protocol is selected for 'ESC/Motor protocol' on the 'Configuration' tab? If it's not 'Brushed' switch it to 'Brushed' and you'll be fine.

ErfanDL, 2017-05-12:
@mikeller thanks for reply. motor configuration is Brushed but not worked. I downgrade to 1.9.0
this problem is in firmware 1.9.0 and above ( 1.10 to 2.0.2 )
and a question: can I use GPS on my CJMCU micro quad for mission planner ?

mikeller, 2017-05-13:
You probably can use GPS on firmware < 2.0. For 2.0, GPS support has had to be removed because there was no space for it.

ErfanDL, 2017-05-13:
@mikeller thanks. Can GPS working with firmware 1.9.0 ?

McGiverGim, 2017-05-15:
There's no firmware 1.9.0. I think remember that the latest was 1.4.2 before 2.0.

ErfanDL, 2017-05-15:
why CJMCU cant support GPS ? my Ublox 7 not worked with CJMCU firmware1.9.0

mikeller, 2017-05-15:
@ErfanDL: Please read my statement from 2 days ago.

closed 2017-05-15 by mikeller

ErfanDL, 2017-05-15:
@mikeller I read. my firmware is not 2.0 or above Im in 1.9.0 :( but GPS not worked

AleksandrNull, 2017-05-31:
Guys, I could confirm that last working version for my CJMCU is 1.14.2. Starting with 2.0.0 and later all of them cause props to start spin around 5-10 seconds after board is booted up. My initial thought was ... hmm.. looks like my mosfet's dying but reverting back to 1.14.2 fixed it. I have brushed with 16k PMW settings in configurator. Dunno what to look into and already tried to look on all commits since tag 1.14.2 but it's a huge amount and, you know, that could be anything. It looks like something initialized on the board which somehow affects PWM output for brushed motor MOSFET control pins.

mikeller, 2017-05-31:
@AleksandrNull: What did you configure as 'ESC Protocol'?

AleksandrNull, 2017-05-31:
Hi Michael, thanks for a reply. For ESC Protocol this board have BRUSHED. I've checked that multiple times.
[image]

Smeat, 2017-05-31:
I have the same problem.
1.14.2 is working flawlessly and with 2.0.2 my motors start spinning after a few seconds (not all just a few of them).
If I try to control the motors manually via the configurator, the connection drops and LED3 is blinking very fast. ESC is set to brushed.
CLI seems to be broken too.

AleksandrNull, 2017-05-31:
CLI is disabled on CJMCU target. Probably because there is not enough space to fit it in.

PR #2831 (2017-06-01): Fix CJMCU pin mapping.
This should fix #2796.
The problem occurred because motor 3 and 4 were mapped to PB6 and PB7, which are used for I2C.

I am no expert on this and don't know if I broke anything else with this.
For now I just restored the functionality from 1.14.2 with PA8 as CH5/AUX1.