cleanflight/cleanflight #2797: Problem magnetometer F4EVO

Bubscanard, 2017-05-12:
Hello,
I am the happy owner of a F4EVO.
I can not operate the magnetometer, in fact the icon remains desperately gray.
The baro is functional, is there a way to unlock the magnetometer?

Thanks in advance,

Cordialy,

McGiverGim, 2017-05-12:
As far as I know, the F4EVO does not have compass (magnetometer). You must add an external one.

http://seriouslypro.com/spracingf4evo

Have you added an external compass?

Bubscanard, 2017-05-12:
Good evening indeed I was wrong.
Besides, as long as I am there I ask myself more questions:
I would like to install a GPS and bluetooth module.
For the ublox GPS module I do not know which one to take, F4EVO is like CC3d or Naze32 ...?
For the model of a model blutooth idem can you advise me?
thanks in advance

McGiverGim, 2017-05-13:
I don't know to much about it. I suppose anything with compass and ublox 8 will be fine. For support please use slack:

https://cleanflight.slack.com/

If this is not an issue, I close it.

Thanks!

closed 2017-05-13 by McGiverGim