cleanflight/cleanflight #2799: Motors spin when fc is rebooted

RaffaeleNachos, 2017-05-13:
Hi, 
Maybe i've found a new issue on cleanflight 2.0.2.
My FC is an SP Racing F3, the ESCs are Racestar RS30A. 
With all kind of protocols such as multishot, oneshot 125 or dshot 300 when the fc is rebooted while connected with usb to cleanflight program it spins the motor at full throttle for 1 second. It is random, not always happens.
Hope this is helpful, downgraded my fc to cleanflight 1.14.2 where all is fine!

tonipalmgren, 2017-05-15:
What is your board's exact model? EVO? NEO? Acro/Deluxe?

I'm having a similar issue on my F3 EVO, but only when I disable or enable blackbox. When disabling, the rear left motor spins. When enabling, the rear left motor spins faster and the front right motor spins noticeably, but not that fast. It's not enough to flip the quad, but might sting if your fingers got in the way.

RaffaeleNachos, 2017-05-16:
Mine is an SP racing F3 Deluxe. I've not tried with the blackbox switch 😂. Yeah always props off...

McGiverGim, 2017-05-16:
Have you flashed the firmware with the "full erase option"? Have you recover the config from a backup dump? If yes, please erase all and config the board from zero. Some fly modes and other things have changed. I had a similar problem and reflashing with erase all solved the problem for me.

RaffaeleNachos, 2017-05-17:
I've flashed with "full erase option", but i've also recovered from dump. I will try to configure the quad from the beginning with all info erased.

lunebleue, 2017-05-17:
Hi, same issue on my F3 EVO that was flashed with full erase option and configured from scratch without using a dump. It has happend many times already when I'm leaving the CLI... "props spin a few turns"...  Now I'm used to it and I keep a hand on the tricopter and the other away from props when I'm saving and leaving the CLI...

hydra, 2017-06-06:
@RaffaeleNachos I think we also need to know exactly what firmware is on your ESCs. 
Can you also give us an update as to whether reconfiguring from scratch helped?

@lunebleue What ESCs are you using?  What ESC protocol are you using?

hydra, 2017-10-21:
closing, no feedback from user.

closed 2017-10-21 by hydra