cleanflight/cleanflight #2800: GPS Ublox 7 not worked on CJMCU

ErfanDL, 2017-05-15:
Hi. I install Ublox 7 on CJMCU but cant worked when I set GPS to Ublox after save and reboot its jumped to NMEA !!! the firmware is version 1.9.0. (the port GPS is enabled and baud set to 57600)

please help

McGiverGim, 2017-05-15:
The F1 CPU's have very few free space, in version 2.0 the GPS has been removed for a lot of targets, so it will not work. You have two options:
- Stay in CF1.x
- Customize your own build to remove some features that you don't need and add the GPS again.

This is not an issue, if you need more support please go to the slack chat: https://cleanflight.slack.com

I close this.

Regards!

closed 2017-05-15 by McGiverGim

ErfanDL, 2017-05-15:
@McGiverGim thanks. but Im in firmware 1.9.0 not 2 or above. my question: is CJMCU microquad support Ublox 7 GPS ?

McGiverGim, 2017-05-15:
I don't know too much about version 1.9. And here you will only find help for issues of version 2.0 or later.

Please try to find help in the slack chat.

Regards!

ErfanDL, 2017-05-15:
@McGiverGim thanks. Can you invite me in Slack ? my email: erfan.shirzad@gmail.com

mikeller, 2017-05-15:
@ErfanDL: You can invite yourself: http://cleanflight.com/slack/ ;-)

ErfanDL, 2017-05-16:
@mikeller Im joined slack support chat but nobody help me ! please build a firmware for me CJMCU that support GPS I need firmware 1.9.0 with GPS