cleanflight/cleanflight #2801: Servo_tilt not working as before

pinkywafer, 2017-05-16:
Before 2.0, Servo_tilt shifted the motor outputs by 2. Now the motor outputs do not shift. If I connect servos to motor outputs 5+6. Only the one at output 5 works as expected, the other does not work with channel assigned or with stabilisation mode.

mikeller, 2017-05-17:
With 2.0 you will have to do the resource assignments for servos yourself. Most likely, the correct assignments will be the same as they were before 2.0 If the servo outputs do not work with your assignment, this is caused by resource conflicts between the motor outputs and the servo outputs. 

Read https://github.com/betaflight/betaflight/wiki/Betaflight-resource-remapping and https://github.com/betaflight/betaflight/wiki/Remapping-Motors-with-Resource-Command-(3.1) for an overview on resource remapping, and try to get it working.

If you can't, somebody in here can probably help you, if you let us know which board you are using, since resources are board specific.

pinkywafer, 2017-05-18:
Thanks, I've got that sorted with manually mapping the resources now.
Jenny

closed 2017-05-18 by pinkywafer