cleanflight/cleanflight #2803: Cant enable Gyro 32khz sampling mode

Bardiaesm, 2017-05-17:
# A - Description

CleanFlight 2.03
SP Racing F3
Cant seem to enable Gyro 32khz sampling mode. When enabled, after rebooting it goes back to before and disable it self. Please watch the video

# B - Steps To Repeat

Watch the attached video

# C - Video Links

https://youtu.be/lcFFgdZcj3I

# D - Forum post links

https://www.rcgroups.com/forums/showthread.php?2889678-CleanFlight-Cant-enable-Gyro-32khz-sampling-mode#post37540943

# E - Version Information
Configurator: 2.0.3

mikeller, 2017-05-17:
Try if `set gyro_lpf = off` in CLI fixes this.

hydra, 2017-05-29:
SP Racing F3 Acro/Deluxe uses I2C, not supported.  Use a board with SPI and a sensor that can support the rate.

closed 2017-05-29 by hydra