cleanflight/cleanflight #2805: im trying to connect my cc32 and the program keep saying: You need to upgrade your firmware

Jeanpillot, 2017-05-18:
# 1. ARE YOU REQUESTING SUPPORT?

-- YES / NO --

# 2. If you answered YES to the above question go here instead:

https://www.rcgroups.com/forums/showthread.php?2249574-Cleanflight-firmware-for-STM32F3-based-FCBs-Check-First-Post-Please%21%21

Alternatively use other forums, slack or IRC for support.

# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

-- replace this line with a summary of the issue/problem/feature

# B - Steps To Repeat

-- replace this line with a list of steps to repeat your problem

# C - Video Links

-- provide a link to a video showing your problem, as applicable --

# D - Forum post links

-- provide links to forum posts if you have discussed this with others --

# E - Version Information

Cleanflight: vX.X.X or commit SHA
Configurator: vX.X.X or commit SHA

# E - Other information

-- provide additional information such as.
-- flight controller make/model
-- config dump + diff
-- esc make/model/config
-- receiver make/model/config
-- transmitter make/model/config
-- schematics
-- diagrams
-- operating system
-- etc.

mikeller, 2017-05-18:
That means that you need to upgrade your firmware. Go to the 'Flash Firmware' tab and follow the instructions there.

McGiverGim, 2017-05-18:
Or if you don't want to upgrade, you must uninstall the configurator, and install manually an older version of it.

fftunes, 2017-05-19:
Just manually install older version of configurator.

No need to delete the current version, and no need to upgrade if you don't want.

closed 2017-05-29 by hydra