cleanflight/cleanflight #2806: virtual current meter not working

pinkywafer, 2017-05-18:
Current meter displays 0.00 both on configurator and on telemetry
Configurator 2.0.3
FW 2.02
Board Naze32 rev5
Updated with full chip erase from 1.12.1 when current meter was working properly. 
previously used: 
`set amperage_meter_scale = 166
set amperage_meter_offset = 60`
tried `set ibata_scale = 166
set ibata_offset = 60`
still nothing.
returned to defaults and still nothing.
full dump below:
```
# Cleanflight / NAZE 2.0.2 Apr 22 2017 / 14:41:48 (4328b13)
name FPV250
resource BEEPER 1 A12
resource MOTOR 1 A08
resource MOTOR 2 A11
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 A02
resource PWM 4 A03
resource PWM 5 A06
resource PWM 6 A07
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A06
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A06
[... 458 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
```

McGiverGim, 2017-05-18:
The Virtual Current Meter was temporary disabled in NAZE32 since 11 Apr:

https://github.com/cleanflight/cleanflight/commit/3d1e755d66c706e0ae92f7698d86f08fd8aa6b4e

I don't know when it will be enabled again.

pinkywafer, 2017-05-18:
Thanks.   Any idea why? If nothing major,  I might do a custom build

McGiverGim, 2017-05-18:
No, no idea.

mikeller, 2017-05-19:
@pinkywafer: My guess is to make the firmware fit into the flash space. F1 boards are always short on space nowadays. Let us know if you need assistance with the custom build.

pinkywafer, 2017-05-19:
Thanks guys.  I first tried a build (based on v2.0.3) with virtual current meter enabled and as suggested, the build was too large.
I've now built again with current meter enabled and HOTT telemetry disabled (as I don't need it.  This is working fine, and I'm happy with the results.

Thanks for the help

mikeller, 2017-05-20:
Good to hear it worked, @pinkywafer. Can you please close the issue?

pinkywafer, 2017-05-20:
Thanks.
Jenny

closed 2017-05-20 by pinkywafer