cleanflight/cleanflight #2808: Speed/default optimized code

ledvinap, 2017-05-19:
All files in F4/F7 and lot of files for F3 are optimized with -Ofast or -O2. Benefit of speed optimization is highly questionable and features are being dropped due to memory limits.

Is there some measurable benefit of optimizations? At least on high gyro rates?

mikeller, 2017-05-20:
The optimisation settings were discussed in the Betaflight slack before they were introduced, and the benefits were that they allowed higher gyro rates in the first place. Unfortunately, the slack history from back then has not been preserved, and there does not seem to be a reference to the performance improvements in GitHub (https://github.com/betaflight/betaflight/pull/1712).

martinbudden, 2017-05-20:
I agree that we could reduce the number of files that are speed optimised. Certainly for F4/F7 we could set the list the same as F3. And there are a number of files that could be removed from the F3 list.

mikeller, 2017-05-20:
Yes. For F4 we took the 'there's enough flash space, let's keep it simple' approach. I think what we should have are 'optimise for size' for F1, and 'selectively optimise for speed' for the rest.

ledvinap, 2017-05-20:
I'm still sceptic about optimization influence, but I can be wrong.
Unrolling loops and more aggressive inlining may get some percent improvement, but nothing extreme. Improving algorithmic complexity is usually way to go.

It would be great to get some profiling results. Some code instrumentation may be worth the effort. Or get someone with SWI trace capability ..

borisbstyle, 2017-05-20:
@ledvinap
I think indeed that unrolled loops give the biggest performance gain. And there are quite some of those where xyz axis calculations are involved

martinbudden, 2017-05-21:
I agree with @ledvinap , "in the large optimisation" (ie good design and algorithms) gets you much more than "in the small" (ie instruction level) optimisation. I think many modules that are currently speed optimised gain little benefit from doing so, and that speed optimisation could be switched off. However there are modules (notably the gyro and pid modules) where that optimisation (and in particular the loop unrolling has benefit).

hydra, 2017-06-06:
this is resolved in master now.  F1 optimization is size, >F1 it's selective now.

closed 2017-06-06 by hydra