cleanflight/cleanflight #2809: Suggestion:  Monitor/Log flight controller voltages

cgapeart, 2017-05-20:
# A - Description

I am debugging a problem where I think either the 5v power rail or the 3.3 v power rail is sagging.  This is not a cleanflight problem - it's probably my rats nest of wires and poor soldering.  

However, I would like to monitor the flight controller voltage separately from the battery voltage, and have that logged  Since I don't use the current or RSSI ADC channels, I am planning on putting in appropriate signal conditioning and tieing the 3.3 v and 5v lines on the FC to those inputs.

My suggestion is to make that a configurable option, and label the data properly both in the MSP protocol and where applicable in other telemetry and logging protocols.  

# B - Steps To Repeat
n/a

# C - Video Links
n/a

# D - Forum post links

n/a

# E - Version Information

Cleanflight: v2.0.2 
Configurator: v2.0.3

# E - Other information

n/a

mikeller, 2017-05-21:
@cgapeart: That is a good suggestion. I think @hydra has got something in the works already that will allow monitoring of multiple voltages concurrently.

hydra, 2017-05-29:
likely logging wouldn't be useful if your 3.3v rail is sagging, if your 3.3v rail *is* sagging then you'd also probably get brownouts on the sdcard or flash chip.

we have to concentrate on features that will benefit all users.  this kind of logging is not such a feature and is the kind of code that you add when designing your own FC to make sure it works and not for general public use.

closed 2017-05-29 by hydra