cleanflight/cleanflight #2822: I2C Errors: 1 in Clean Flight

Talha909, 2017-05-28:
I am using F3 10DOF. There is error I2C Errors: 1. If i disable the barometer then the error gone. I am on new version of the clean flight This thing is shown when i use the status command. 

$M>tARM;AIR MODE;ANTI GRAVITY;ANGLE;HORIZON;HEADFREE;BARO;MAG;HEADADJ;GPS HOME;GPS HOLD;FAILSAFE;BEEPER;BLACKBOX;BLACKBOX ERASE (>30s);FPV ANGLE MIX;OSD SW;´$M>ü	h$M>n¸Ñ
CLI

# status
System Uptime: 475 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 1, config size: 1898, max available config: 4096
CPU:9%, cycle time: 504, GYRO rate: 1984, RX rate: 49, System rate: 9

#

mikeller, 2017-05-28:
It's quite normal that you get I2C errors during the initialisation of devices on the I2C bus, no reason to worry. I would assume that your board has the barometer on I2C?

Talha909, 2017-05-28:
I havd F3 10DOF. All the thing build in.  But when i flashed the INAV then their is not error. After that I again flashed the Clean Flight. I got the error again.  Why INAV is not shown error.

mikeller, 2017-05-28:
My guess would be that the initialisation sequence in iNav is different, causing the error not to show up.

So is there a problem caused by this error? If you are worried about it, then maybe you should switch to using iNav.

Talha909, 2017-05-28:
I like Clean Flight.  But error make me confused.  Thanks for your support.

closed 2017-05-29 by hydra

Talha909, 2017-05-30:
Hydra any comments on my error.