cleanflight/cleanflight #2828: Motors don't start after required updating to latest CleanFlight version

Phipos, 2017-05-29:
# 1. ARE YOU REQUESTING SUPPORT?

-- YES --

# 2. If you answered YES to the above question go here instead:

https://www.rcgroups.com/forums/showthread.php?2249574-Cleanflight-firmware-for-STM32F3-based-FCBs-Check-First-Post-Please%21%21

Alternatively use other forums, slack or IRC for support.

# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

I am using the latest CleanFlight software version.
I couldn't connect my NAZE32 10DOF Rev6 to CleanFlight because my firmware was obsolete.
CleanFlight request me to use an upgraded NAZE32 firmware.
I tried to flash all different versions of available firmwares, but each time the same problem occurs :
The motors won't run at all, either with the motors tab, or with the RC transmitter. and I am sure that the motors are armed.
I have tried several recent versions of CleanFlight (and BetaFlight) with no success.
However, when using another contoller utility (BaseFlight), everything work fine, but the settings are too different to adjust.
Is this issue is already known ?

# B - Steps To Repeat

Each time the same problem occurs, on two different NAZE32 controllers.
The only way to have the motors running is to flash the NAZE32 with another software (such as BaseFlight), which is not my prefered choice.

# C - Video Links

-- provide a link to a video showing your problem, as applicable --
Not available.

# D - Forum post links

-- provide links to forum posts if you have discussed this with others --
Not so far...

# E - Version Information

Cleanflight: v2.0.2
Configurator: v2.0.3

# E - Other information

-- flight controller make/model : NAZE32 10DOF Rev6
-- config dump + diff : N.A.
-- esc make/model/config : SIMONK 20A
-- receiver make/model/config : REDCON CM703, 7 channels (from Banggood)
-- transmitter make/model/config : SPEKTRUM DX7
-- schematics : N.A.
-- diagrams : N.A.
-- operating system : Windows 10, 64 bits
-- etc.

mikeller, 2017-05-30:
Can you please provide a CLI `diff`?

hydra, 2017-06-02:
likely just a mis-configuration error relating to your ESC output configuration.  CF uses oneshot by default now and perhaps you need PWM?

closing, no response from OP.  Please use cleanflight slack or forums for further support.

closed 2017-06-02 by hydra