cleanflight/cleanflight #2833: IBUS telemetry - Change of sent parameters label:idea

Hrumque, 2017-06-01:
It is very cool that for Flysky apparatus ibus telemetry works. (It would be helpful to describe the sharing of one usart for the ibus receiver and the ibus telemetry transmission)

In classical ibus devices there are (so far) hardware sensors:
- the power supply to the receiver (5v, unless you made the receiver mod)
- battery voltage
- temperature
- RPM (optical or magnetic)
Sensors of one type can be physically several, then we have measurements such as "external voltage 1", "external voltage 2"

Cleanflight very nicely replaces these hardware sensors, sending the same data but created in the flight controller. And here it would be useful to change something: Instead of one parameter being sent (eg Vbat voltage) - it would be possible to send several parameters of the same type, with configurable assignment to the value measured by FC:
- Since ibus does not have (so far) a current sensor - you can send the measured current, eg as a second voltage (just make an appointment that eg "external voltage 2" = 56.2V this means 56.2A)
- Since returning "rotation" that changes just like a throttle (eg 1000-2000) does not make sense - would not it be better to send the calculated mA / measured mA in this parameter?

- It would be useful to return altitude (whether it's barometer or gps), which is probably more important and functional than the temperature (to which most users do not have a sensor). https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/ibus_shared.c#L134 change   "value = gyroGetTemperature() * 10;" to something with "attitude" variable?

McGiverGim, 2017-06-02:
INAV has some of this features, and some others in progress (https://github.com/iNavFlight/inav/pull/1495):

You can see at least altitude and current:

+sensor 4 is of type S85,
 +sensor 5 is of type ACC_Z,
 +sensor 6 is of type CURRENT,
 +sensor 7 is of type ALT,
 +sensor 8 is of type HEADING,
 +sensor 9 is of type DIST,
 +sensor 10 is of type COG,
 +sensor 10 is of type GALT,
 +sensor 12 is of type GPS_LON,
 +sensor 13 is of type GPS_LAT,
 +sensor 14 is of type ACC_X,
 +sensor 15 is of type ACC_Y, 
 +sensor 16 is of type SPEED.

Hrumque, 2017-06-02:
> INAV has some of this features, and some others in progress (iNavFlight#1495):

In the case of the INAV project - I lack one simple information, visible in a readable place. What version of Cleanflight does the given version of inav rely on?

hydra, 2017-06-06:
In principle I'm now against mis-using telemetry values for purposes other than their original intended use.  Using a 'voltage' for something other than a voltage is bad.

It creates a big problem in terms of documentation, user expectation, etc.  We had this with the FrSky telemetry and it's a situation I'm not willing to repeat.

Better solution is to request the the Flysky guys update their telemetry protocol and firmware and then we can add support in CF for the things it doesn't currently support.  e.g. ask them to add support for 'current, altitude, etc'.

If I'm mis-understanding your requirements please explain.

hydra, 2017-06-06:
@unitware @blckmn  Do you guys have any thoughts on this ?

unitware, 2017-06-07:
Maybe check with @qba667 so that the modded flysky firmware goes hand in hand with the changes. Then new sensor types can be added on both sides. 

Of course if anyone has contact with flysky proper then that would be even better.

(I'm working on other stuff and use OSD as my main flight-info channel at the moment so I'm not that involved in ibus-telemetry anymore.)

qba667, 2017-06-07:
@hydra @unitware
Please use definitions from:
https://github.com/qba667/FlySkyI6/blob/master/source/source/ibustelemetry.h
I have tried to contact FlySky representative but no real response, only evasive declarations like "must be consulted with xyz". Im my definition the IBUS_MEAS_TYPE_ALT is using different id than FlySky but this value in I6X and in 10 are calculated from pressure and temperature so there is no sensor data wuth this id send by air. For me it was easier to mange additionally allocated memory when sensors ids are consecutive.
The concept of supported sensors, mapping and discovery can be taken from
https://github.com/qba667/MAVLinkToIbus/tree/master/MAVLinkToIBUS
(it must be also updated to use definitions from ibustelemetry.h).
This implementation can be just reused - there is one critical part in my UART handling it is bidirectional but half duplex mode where I control rx/tx ability, and some precision timing is also needed.

hydra, 2017-10-21:
any progress in the communications with FlySky @qba667 ?  Perhaps try again?

qba667, 2017-10-23:
@hydra I have no connections to FlySky. But I have already added my comment and link to definitions and data types used.
My FW used add sensor types used by FlySky with only one minor difference:
IBUS_MEAS_TYPE_ALT_FLYSKY	has different id than my IBUS_MEAS_TYPE_ALT.
I will try to implement sensors missing in cleanflight - there is one implementation from @hali9 but far to complex.

closed 2020-05-24 by stale[bot]