cleanflight/cleanflight #2834: CH11-16 (Aux 7-12) not responding on Naze32

deltaflyer4747, 2017-06-03:
Naze32 connected to SBUS receiver via UART2 does not respond (not even in "Receiver" tab) to channels 11-16 (AUX7-12), responds to other chans properly. Clean install, chip erased, configured only 

Verified that this is not HW issue - checked with osciloscope on SBUS and iNav, changes are visible and responding in iNav.

Cleanflight: v2.0.2
Configurator: v2.0.3
Naze32 rev.6 Full

mikeller, 2017-06-03:
Try `set max_aux_channels = 12` in CLI.

deltaflyer4747, 2017-06-03:
That did the trick, thank you.

Could this be set as default for SBUS recvs ?

McGiverGim, 2017-06-03:
The processing of aux channels is a process that consumes CPU. For F1 CPUs (like Naze) the developers let only ten channels to use the CPU for better response in flight.

deltaflyer4747, 2017-06-03:
Makes perfect sense. Thank You!
(will close in 24hrs in case anyone wants to add anything)

closed 2017-06-04 by deltaflyer4747