cleanflight/cleanflight #2844: CleanFlight not sending VFAS over Smart Port on SP Evo Brushed

chrismcna, 2017-06-06:
# A Description
CleanFlight not sending VFAS over Smart Port on SP Evo Brushed.

I have an SP Evo Brushed FC with built in battery monitoring running Cleanflight 2.0.2, connected to XSR Frsky taranis X9D Plus. Everything is working as expected, including smart port telemetry to taranis excluding VFAS, Cleanflight configurator shows the correct battery voltage.

# B - Steps To Repeat

Buy SP Evo Brushed, connect to taranis using a Frsky X Receiever and setup Smart Port telemetry.
VFAS is not being sent by Cleanflight

# E - Version Information

Cleanflight: v2.0.2 

# E - Other information
I have fixed this issue by downloading source, changing Telemetry/SmartPort.c line 619.
i changed this Smart Port sensor to the block of code below.
basically my issue is caused because (getBatteryCellCount() > 0) = false.
So i have changed the code to retain the current functionality but if the cell count is  <= 0 then send the voltage anyway

<code>
            case FSSP_DATAID_VFAS       :
                if (batteryConfig()->voltageMeterSource != VOLTAGE_METER_NONE) {
                    uint16_t vfasVoltage;
                    if (getBatteryCellCount() > 0 && telemetryConfig()->report_cell_voltage) {
                        vfasVoltage = getBatteryVoltage() / getBatteryCellCount();
                    } else {
                        vfasVoltage = getBatteryVoltage();
                    }
                    smartPortSendPackage(id, vfasVoltage * 10); // given in 0.1V, convert to volts
                    smartPortHasRequest = 0;
                }
                break;
</code>

chrismcna, 2017-06-06:
it looks like the editor has removed the Crlf

case FSSP_DATAID_VFAS : 
	if (batteryConfig()->voltageMeterSource != VOLTAGE_METER_NONE) { 
		uint16_t vfasVoltage; 
		if (getBatteryCellCount() > 0 && telemetryConfig()->report_cell_voltage) 
		{ 
			vfasVoltage = getBatteryVoltage() / getBatteryCellCount(); 
		} 
		else 
		{ 
			vfasVoltage = getBatteryVoltage(); 
		} 
		smartPortSendPackage(id, vfasVoltage * 10); // given in 0.1V, convert to volts smartPortHasRequest = 0; 
	} 
break;

mikeller, 2017-06-06:
Why do you want your voltage to be reported when there is no battery attached to the quad (i.e. `getBatteryCellCount() == 0`)? If you need this change, then I suspect your configuration is faulty, and the battery is not recognised.

chrismcna, 2017-06-06:
there is a battery

chrismcna, 2017-06-06:
for some reason the BatteryCellCount is 0

chrismcna, 2017-06-06:
Cleanflight configurator displays the correct battery voltage but doesn't send it via Smart Port, until i changed the code, recompiled and flashed the flight controller

mikeller, 2017-06-06:
Well if `getBatteryCellCount()` returns 0 then the flight controller thinks there is no battery, and battery monitoring will not work.

I suggest you check your configuration, and make sure that `bat_detect_thresh` (for 3.1.7) is below your battery's voltage.

chrismcna, 2017-06-06:
to be honest, i changed the if statement (620) to be "if (true == true)" to get the VFAS working on the FC.
Then i decided that as I'm having this issue then i bet other people are, so i asummed that out of the two conditions in the statement

1: "batteryConfig()->voltageMeterSource" - i am getting voltage in configurator and its correct so i asummed that my Voltage Meter Source must be okay

2: "getBatteryCellCount() > 0" i am unsure where is can see this value in Cleanflight configurator, so asummed this must be the issue.

I now realise that i should test this and find out which condition is causing my issue.

chrismcna, 2017-06-06:
Under the battery section in Cleanflight configurator battery connected is always "no", no matter if the battery is connected or not. 

I will check "bat_detect_thresh" when i get home later

chrismcna, 2017-06-06:
how do i check "bat_detect_thresh"? it isnt mentioned in the Cli.md

https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md

mikeller, 2017-06-06:
In CLI. Unfortunately the documentation tends to get woefully outdated.

chrismcna, 2017-06-06:
oh okay, Ive been using that a lot to look for the commands :( atleast i know now

so i guess its just "get bat_detect_thresh"?

hydra, 2017-06-06:
just configure the battery threshold correctly for your battery, the default in CF 2.0 is 5.5v which is probably when the cell count is 0 if you're using a 1S brushed quad.  set it to about 3v (new default in CF 2.1) and you're done.  If you need further assistance on setting this please use the cleanflight slack channel.

closed 2017-06-06 by hydra

chrismcna, 2017-06-07:
I am using an 1S battery and issuing the command "set bat_detect_thresh = 30" (3v) fixed my issue. 
Thanks for the help, 
sorry for opening an issue when it was a configuration fault

hydra, 2017-06-07:
no worries, glad it was resolved for you.