cleanflight/cleanflight #2854: 'dump' not showing resource settings

pinkywafer, 2017-06-09:
I have changed my resource assignments to do the 'shift by two' like the old camstab did automatically, and this is now working properly, but the changes do not appear in a `dump`.  they do appear if i type `resources`
`dump` gives me the following resources section:
```resource BEEPER 1 A12
resource MOTOR 1 A08
resource MOTOR 2 A11
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 A02
resource PWM 4 A03
resource PWM 5 A06
resource PWM 6 A07
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A06
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
```
which is not what I have set
`resource` gives the following (correct version):
```
resource BEEPER 1 A12
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource SERVO 1 A08
resource SERVO 2 A11
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 A02
resource PWM 4 A03
resource PWM 5 A06
resource PWM 6 A07
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A06
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
```
this is on a NAZE32 rev5 with a custom build of cleanflight - virtual current meter enabled and hott telemetry disabled

mikeller, 2017-06-11:
Have you done a `save` after configuring the resources, before doing the `dump`? Resource assignments only become active on save / reboot.

pinkywafer, 2017-06-11:
Yes,

I set resources with:
```
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource SERVO 1 A08
resource SERVO 2 A11
save
```
after reboot, if I do cli command `resource` the returned values are those I've configured, but if I use `dump` the defaults are displayed.  The resources are being used as programmed, but not displayed correctly in the `dump`

I've been having a look at the printResource function in cli.c and believe the issue may be somewhere in the loading of the values.  I tried changing line 2775 to  `if (dumpMask & DO_DIFF || dumpMask & SHOW_DEFAULTS || dumpMask & DUMP_MASTER){` to try to see what happened.
with this, if I do a `dump` the correct values are displayed in dump and `resource`, BUT now if `resource` is called first (ie before `dump`), no values are returned.  To test further, I removed the `HIDE_UNUSED` from the `resource` command, and now when `resource` is called first, all resources show NONE for assignments, `dump` shows the correct assignments, and `resource` shows the correct assignments AFTER `dump`

pinkywafer, 2017-06-11:
ok, i'm looking at the printResource function
```
or (unsigned int i = 0; i < ARRAYLEN(resourceTable); i++) {
        const char* owner = ownerNames[resourceTable[i].owner];
        const void *currentConfig;
        const void *defaultConfig;
        if (dumpMask & DO_DIFF || dumpMask & SHOW_DEFAULTS) {
            const pgRegistry_t* pg = pgFind(resourceTable[i].pgn);
            currentConfig = pg->copy;
            defaultConfig = pg->address;
        } else { // Not guaranteed to have initialised default configs in this case
            currentConfig = pgFind(resourceTable[i].pgn)->address;
            defaultConfig = currentConfig;
        }
...
```
If I change to:
```
...
        } else { // Not guaranteed to have initialised default configs in this case
            currentConfig = pgFind(resourceTable[i].pgn)->copy;
            defaultConfig = currentConfig;
        } 
...
```
`dump` now returns the correct resources.
but if `resource` is called first, it doesn't return anything - I've tried removing the HIDE_UNUSED dumpMask from cliResource so I can see what's going on, and in this case, all resources come up as NONE. 
but if `resource` is called after dump, it returns the correct resources.
What am I missing?

mikeller, 2017-06-13:
I think the bit that you were missing is that `printResource()` can be called in 2 different situations: When the config has not been backed up into `copy` (when it is called during CLI `resource`), and when it has been backed up into `copy` (during CLI `diff` / `dump`). When I implemented the current form of `printResource()`, I thought I should be able to derive which one it was from the context. But turns out I can't.

pinkywafer, 2017-06-13:
ok, I see where it is now, thanks - should have spotted that before!

If you like I'll send a PR from mine

pinkywafer, 2017-06-13:
closed PR due to errors.

@mikeller fix in betaflight#3276 is much neater

pinkywafer, 2017-06-13:
closed by mistake!

closed 2017-06-14 by borisbstyle