cleanflight/cleanflight #2857: mspProcessOutCommand / mspProcessInCommand

ledvinap, 2017-06-09:
msp processing is split into ProcessInCommand / ProcessOutCommand.
As implemented, it is impossible to pass argument to ProcessOutCommand (and return something other than error from ProcessInCommand).

Some commands that needs special-casing just to handle input arguments (mspFcWpCommand, mspFcDataFlashReadCommand)

CF 1.x used `int mspServerCommandHandler(mspPacket_t *cmd, mspPacket_t *reply)` to handle all commands (one large switch) https://github.com/cleanflight/cleanflight/blob/master_v1.x/src/main/fc/msp_server_fc.c#L519

Another option is to pass input buffer to <x>ProcessOutCommand, so that special cases can be merged ..

(I want to implement paramater to MSP_BOXNAMES, MSP_BOXIDS, https://github.com/cleanflight/cleanflight/pull/2856. Without change it will be too ugly)

ledvinap, 2017-06-09:
**Vote:** merge ProcessInCommand / ProcessOutCommand into single function

ledvinap, 2017-06-09:
**Vote:**: Pass `mspPacket_t *cmd` to ProcessOutCommand

ledvinap, 2017-06-09:
@martinbudden : What about functions that return some info (OutCommand) but need (optional) parameter to do so?

martinbudden, 2017-06-09:
I would make them special functions called from within `mspFcProcessCommand`.

Generally speaking, as a development methodology, I think it is best not to change the framework when adding something new - rather work within the existing framework and then, if it looks really ugly refactor. I think this is generally better than taking the approach "this is going to be really ugly, so I better refactor as part of doing it".

So I'm not 100% against refactoring this area, but I'd like to see if it really does turn out ugly before we make that decision.

ledvinap, 2017-06-10:
@martinbudden : What about this: https://github.com/cleanflight/cleanflight/pull/2856/files#diff-ffa49532e0a35299ef3e307058236e8eR1380

I would also eventually change other handlers to return MSP_RESULT_CMDUNKNOWN, that way handler can return error (for example 'command can't be executed now') without falling through all other handlers

martinbudden, 2017-06-11:
@ledvinap , yes I'm happy with that approach.

hydra, 2017-10-21:
@martinbudden @ledvinap do we still want to do this? does either of you have time?

hydra, 2020-05-17:
Closing, the code has changed significantly with the merge of BF 4.2 into `master`.  Please re-open if there's still concerns.

closed 2020-05-17 by hydra