cleanflight/cleanflight #2861: BARO MS5611 BUG in SPI mode (wrong register called for pressure reading)

djspark1083, 2017-06-15:
# A - Description

-- MS5611 barometer in SPI mode always displays 0 as altitude.  Found the bug in /drivers/barometer_ms5611.c file, under the function "static void ms5611_start_up(void)".  It should call CMD_ADC_D1 instead of CMD_ADC_D2.  D2 is for temperature.  I2C section is ok.

# B - Steps To Repeat

-- just need to have an SPI baro MS5611 (or 5607)

# C - Video Links

-- none

# D - Forum post links

-- none

# E - Version Information

Cleanflight: v2.1.0
Configurator: v2.0.3

# E - Other information

static void ms5611_start_up(void)
{
#ifdef USE_BARO_SPI_MS5611
    ms5611SpiWriteCommand(CMD_ADC_CONV + CMD_ADC_D2 + ms5611_osr, 1); // D2 (temperature) conversion start!
#else
    i2cWrite(BARO_I2C_INSTANCE, MS5611_ADDR, CMD_ADC_CONV + CMD_ADC_D1 + ms5611_osr, 1); // D1 (pressure) conversion start!
#endif
}

mikeller, 2017-06-15:
Good find! Would you be able to submit a pull request to fix this?

djspark1083, 2017-06-16:
Well, i'm still debugging.  Still getting 0.00 as altitude.  There might be other thing to fix besides that.

hydra, 2017-06-17:
@djspark1083 any update?

djspark1083, 2017-06-17:
Yes i also found other mistakes in spi write and also spi read.  

ms5611SpiWriteCommand() was doing two transfers.  Only one is needed because there is no data to send, just the register address.  No mask is needed for reg.  i removed "data" parameter in the function.

ms5611SpiReadCommand() was also mistakenly using a mask for reg.  That makes it impossible to read the adc.  just removed it.  

With all these changes the MS5611 in spi mode is now working.  The two functions should be as follows:

bool ms5611SpiWriteCommand(uint8_t reg)
{
    ENABLE_MS5611;
    spiTransferByte(MS5611_SPI_INSTANCE, reg);
    DISABLE_MS5611;

    return true;
}

bool ms5611SpiReadCommand(uint8_t reg, uint8_t length, uint8_t *data)
{
    ENABLE_MS5611;
    spiTransferByte(MS5611_SPI_INSTANCE, reg);
    spiTransfer(MS5611_SPI_INSTANCE, data, NULL, length);
    DISABLE_MS5611;

    return true;
}

hydra, 2017-10-21:
this looks like it's fixed in master now, if not please open a new issue and reference this one.

closed 2017-10-21 by hydra