cleanflight/cleanflight #2862: Can't open with desktop shortcut.

thepiper92, 2017-06-16:
Hello, this is a rather minor issue, but I am not sure why it is happening. I initially had Cleanflight and used a desktop shortcut to open. This worked fine, but now all the shortcut does is open a blank Chrome page. I've tried reinstalling it as well. I'm using Windows 10, newest versions of Windows, Chrome and Cleanflight. I also have Betaflight, MWOSD and BLHeli installed, and all have the same issue. I had to purchase MWOSD, so I am concerned about deleting profiles and so forth in Chrome. I have never bought an app before, thus I do not know if removing all traces of Chrome will result in the store asking me to pay again.

mikeller, 2017-06-16:
Try uninstalling Chrome and reinstalling it. Purchased apps should restore just fine, since they are linked to your Google account.

thepiper92, 2017-06-16:
I shall see how it goes, thanks!

closed 2017-06-17 by hydra