cleanflight/cleanflight #2868: Attempt to port cleanflight on "STM32F207XX" fails

Mehedi1997, 2017-06-21:
Hi,
       I am trying to compile the cleanflight 2.1.0 for my stm32f207vct6 board since I have couple of them. I think it will be great to be able to do it because stm32f2xx series MCU is capable of running at 120Mhz and has a lot more resource compared to F1.
I'm using eclipse and latest gnu arm tools and configured them as per suggested by "Building in Eclipse.md" and able to compile the hex files for default targets. 

>>Now I have written  the target dependent drivers for this F2XX series MCU and included them in the supplied makefile along with cleanflight(I don't know how to create a new makefile) . I also added a new target name and added  links for the  StdPerphDriver for stm32f2xx and other sources as it was for F1, F3....F7 targets . along with the StdPerphDriver for stm32f2xx. But it throw the following error massage:

*************************************************************************************************
20:35:19 **** Incremental Build of configuration Default for project cleanflight ****
make TARGET=STM32F207 
c:/program files/gnu mcu eclipse/build tools/2.9-20170607-0922/bin/sh: git: not found
File not found - *.c
make -j ./obj/cleanflight_2.1.0_STM32F207.hex
make[1]: Entering directory 'C:/Users/Benzene/git/cleanflight'
c:/program files/gnu mcu eclipse/build tools/2.9-20170607-0922/bin/sh: git: not found
File not found - *.c
make[1]: *** No rule to make target 'lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F2xx/stm32f2xx_conf.h lib/main/STM32F2xx_StdPeriph_Driver/inc/stm32f2xx_dma.h', needed by 'obj/main/STM32F207/drivers/dma.o'.  Stop.
make[1]: *** Waiting for unfinished jobs....
%% startup_stm32f2xx.s 
%% bus_i2c_stm32f207xx.c 
%% adc_stm32f207xx.c 

make[1]: Leaving directory 'C:/Users/Benzene/git/cleanflight'
make: *** [hex] Error 2
Makefile:1569: recipe for target 'hex' failed

20:35:24 Build Finished (took 5s.110ms)
*************************************************************************************

....hope somone will shade some light on the spot.
``

mikeller, 2017-06-21:
The first error seems to be that make cannot find `git`. Check your path.

Further errors suggest that there is something wrong with the library pathing / includes in your Makefile.

Maybe, if you point somebody at your branch, people can better help you by trying to build what you are working on.

Mehedi1997, 2017-06-22:
Thanks mikller for quick reply, 
the first error was caused by not having git.exe installed on my pc. it was compilng the default targets though. now I have it installed so the first error is gone. 

     all my arm programming knowledge was based on keil and windows so I am a real noob about 'makefile'.
So possibly the mystery of second error is lying inside the makefile. 

I would upload modified repository as soon as possible(3-4 days) in my branch and will be very happy to have someone to look into that....

Mehedi1997, 2017-06-26:
After trying a few days and night  I am now able to compile a hex file for my target  only if  I  delete everything inside the root->obj->main folder and click the hammer icon then it will compile a hex file but every second time I try to compile it again it will throw the following error massage:

make[1]: Leaving directory 'C:/Users/Benzene/git/Cleanflight-stm32f2xx'
make[1]: *** No rule to make target 'lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F2xx/stm32f2xx_conf.h lib/main/STM32F2xx_StdPeriph_Driver/inc/stm32f2xx_dma.h', needed by 'obj/main/STM32F207/drivers/accgyro/accgyro_mpu.o'.  Stop.
make: *** [hex] Error 2
Makefile:1524: recipe for target 'hex' failed
00:04:38 Build Finished (took 8s.192ms)

Now if I upload the hex file through usb DFU tools then start the board without gyro/acc 
(I forgot them in my University) and the LED0 blinks in short...short....short....long...long sequence. 
and nothing else works.

After realizing that it is stuck due to lack of gyro  I attached the following two line in the (root)->src->main->target->STM32F207->target.h: #define USE_FAKE_ACC #define USE_FAKE_GYRO, now the  LED0 blinks with 5 very short pulses and then continuously blinks with 0.5s period.
 I have probed the motors output with my DIY oscilloscope and they all are giving 1ms pwm with approximately 14% duty-cycle. I am using vcp feature, The computer can detect an usb device named CF-F2(Which is the usbd string I defined in target.h file ) but can't find a driver for it.

Now I'm trying to connect the chrome configurator through usb-ftdi module with uart ports but there is no response from the board. I don't know what to do because I never used cleanflight before.

I have uploaded all the codes in my git-hub repository CF-F2 : https://github.com/Mehedi1997/Cleanflight-stm32f2xx/tree/CF-F2.

mikeller, 2017-06-26:
Hi @Mehedi1997.

Try using the command line for building, it's probably something in your IDE that's not set up correctly.

> I have probed the motors output with my DIY oscilloscope and they all are giving 1ms pwm with approximately 14% duty-cycle.

That sounds about right. The default motor protocol is OneShot125, which sends 125us pulses when idle. ('duty cycle' is not the right way to look at it, since it's only pulse length that controls the ESC output.

Try this to get your drivers fixed: http://www.dronetrest.com/t/fix-any-stm32-dfu-drivers-issues-when-flashing-betaflight-cleanflight-firmware/3603

Mehedi1997, 2017-06-26:
Thank you very much!!! @mikeller  this is the best gift for me on this "EID-day" the solution you have given to my problem!! such an awesome guy you are! 
now everything  is working as expected! I am connected to the GUI. I cannot believe that I have done it! 
I have turned my stock Dev-board into a new born flight controller by modifying thousand lines of source codes.!

mikeller, 2017-06-26:
Happy Eid!

hydra, 2017-07-04:
@Mehedi1997 do you have a link to your repo?  Well done on getting things working too!

McGiverGim, 2017-07-04:
This seems the repo https://github.com/Mehedi1997/Cleanflight-stm32f2xx

But I don't know if it's up to date...

Mehedi1997, 2017-07-16:
I am sorry @hydra for being too late to respond. But I was helpless because accidentally I lost my laptop, so lost all the codes and stuffs.
I had to re-do most part of my work as my git repo was not up to date.
In the mean time in my university new semester has begun.
But I have managed to do it better than before . This is the link to my updated repo: https://github.com/Mehedi1997/Cleanflight-stm32f2xx .

hydra, 2017-10-21:
Thanks for the link.

We're deprecating F1 support. F2 support is not required at this time, especially as the F2 doesn't have an FPU.

Thanks for sharing the code though and we hope you had fun.

closed 2017-10-21 by hydra