cleanflight/cleanflight #2870: SP Racing F4evo / external Mag not recognized

trekki1978, 2017-06-22:
Hi,
I'm using the F4evo FC. I wanted to exchange a GPS module w/o compass to a version with integrated compass. GPS works fine but when I connect the compass to the I2C port it is not detected and barometer seems to be offline. After disconnecting the compass and rebooting barometer is online again.
Does someone else also have this issue? How can I find a solution?

Thank you and kind regards
Marco

trekki1978, 2017-06-23:
Hi, mache the issue #2871 is the same.

trekki1978, 2017-06-24:
Hi again,
seems that issue #2871 does not show a solution for my issue. I need to reopen it.
I still can not connect my external magnetometer over I2C port to the FC without loosing the barometer.
Is there maybe a problem with the assignment of the ports?
Thank you
Marco

hydra, 2017-07-04:
does the firmware have support for the mag you're trying to use?  what mag are you trying to use?

#2871 is not related since the F4EVO has i2c on by default because it has a Baro connected via I2C.

trekki1978, 2017-07-10:
Hi, the mag should be HMC5983 (on the chip "A983" is printed). In the meantime I tried an older version GPS/compass module and it works. Atthe moment I'm waiting for a new module as I believe the current is broken. I will update and/or close this issue as soon as I know something more.
Thank you so far.

trekki1978, 2017-07-23:
Hi again, I will close this issue because also the replacement GPS/compass module doesn't work.
Now I tried a separate magnetometer (HMC5983 GY-282). If this is connected to I2C the same happens: no MAG and no BARO. Looks like the controller or firmware has a problem with this MAG/compass.
I will have a look for another module.

closed 2017-07-23 by trekki1978