cleanflight/cleanflight #2871: I2C on omnibus f4 seems not working

onliner10, 2017-06-22:
# Description
When plugging in i2c sensors (e.g. BMP280, or hmc5883l) they are not being detected anywhere, neither in cli status, nor in gui. When browsing the code i encountered following `lines`
```
#define I2C2_SCL                NONE // PB10, shared with UART3TX
#define I2C2_SDA                NONE // PB11, shared with UART3RX
```

Does it mean that I2C is disabled? Can i somehow disable UART3 and enable I2C  back?

# Steps To Repeat
Try to use I2C sensor on omnibus f4 board

# Version Information

Cleanflight: v2.0.3
Configurator: v2.0.3

McGiverGim, 2017-06-23:
I've it working, modifying the target.h and recompiling, but I think it is deactivated to let you assign the pins with the `resource` command. You must free the pins for the UART3 and assign them to the I2C if you prefer I2C over UART3.

mikeller, 2017-06-23:
@McGiverGim is correct, unfortunately at the moment pins can only be assigned to either I2C or an UART in the firmware, and changing the assignment requires a recompile.

There is work under way to make these assignments changeable in the configuration, but it is not quite ready yet. Hang in there!

onliner10, 2017-06-23:
Ok, I've recompiled cleanfligt trying two things, while each time setting correct pins in I2C_DEV2:

- completely commenting out USE_UART3 and following assignments
- setting UART3 pins to NONE

The two solutions above caused three i2c errors to be displayed. Since I think my wiring is quite correct, i thought maybe it is because of not having pullups on i2c line. I browsed through cleanflight code and found I2C_USE_PULLUPS declaration in some other (but similar) target. Set it, recompiled and... those three errors are still present. I'm not sure whether I'd ever bring those BMP280 and HMC5883L to work. Maybe you guys can suggest me something ? Can it be because I first powered for a while this 3V bmp280 module with 5v and something got "on fire"? Can I possibly somehow see some logs, or anyway figure out what those 3 errors are and what do they mean?

McGiverGim, 2017-06-24:
In the latest master, I've only added the pins to the I2C definition:
```
#define I2C2_SCL                PB10 // PB10, shared with UART3TX
#define I2C2_SDA                PB11 // PB11, shared with UART3RX
```
And it worked for me, without removing the UART3 definitions. Try it.

onliner10, 2017-06-24:
Unfortunately, still doesn't work.. Tried iNav and for some reason it works out of the box..

mikeller, 2017-06-25:
@onliner10: To make sure the problems aren't caused by an old config sticking around, do a 'Full chip erase' or a `defaults`.

Mehedi1997, 2017-06-28:
@onliner10, 
 >The two solutions above caused three i2c errors to be displayed.
this sounds like the i2c device is working.

have you defined the  i2c instance for your Baro? i.e added the following line
#define BARO_I2C_INSTANCE		(I2CDEV_2)

mjaniszew, 2017-07-06:
Hi @onliner10 ! I've stumbled on exact same issues, but for Flip32/AIRBOTF4 FC. I've spent days of checking out betaflight/cleanflight/iNav and trying to figure out what the heck is going on with my BMP280 not working with FC and these softs, while it worked perfectly with Arduino, absolutely confused as total RC/drones newbie. At the end it came out that you need to do two things as suggested by previous commenters:
- configure I2C SCL and SDA pins to let FC know that you're be using them to connect I2C device. Disabling UART3 is not necessary (at least it wasn't for me)
- add support of device you want in your FC target, in that case probably adding `#define USE_BARO_BMP280` into `OMNIBUSF4/target.h` in proper section (pay attention that it's already there, but in your case only for OMNIBUSF4SD, so you need to add it in proper section if you're using standard OMNIBUSF4 version)
- and of course build desired target, and flash flight controller with version you've built (I've used recent master branch, not 2.0.3 because it contains changes regarding totally configurable I2C rework)

So if you're still facing those issues and want to use cleanflight instead of iNav (it works out of the box, because it automatically detects pins for I2C shared with UART3 as far I figured out from the code, which had to be manually configured for cleanflight/betaflight). For the reference you can take a look on changes I had to make in cleanflight for myself in order to get this working on my FC (Flip 32 F4 - AIRBOTF4 target): https://github.com/mjaniszew/cleanflight/commit/91ab51f81cce8017c2e089a02fe61ae8efe8b458?diff=unified

Also it's worth of mentioning that if you actually connected BMP280 meant to work with 3,3V into 5V output, you most probably killed it and it won't work anymore (I've killed my first BMP280 in the same way and had to buy new one). You have to provide 3,3V to it somehow: via PBC, via FC provides 3,3v output, or use any kind of Voltage Regulator. It's best to use Arduino to test these things at first place (eg. buy cheapest clone for couple of bucks).

Hope it helps, or at least it will help someone else facing same issue and trying to get exact same newbie path I've gone through! ;)

hydra, 2017-10-21:
thanks for the info, users requiring specific sensor support can compile a customized version as per the documentation and use the resource command to allocate the pins to I2C.

PR's welcomes to add support to targets.  Be aware that non-native sensor support is often removed from targets due to flash size issues.  This is less of an issue on F4 targets though.

closed 2017-10-21 by hydra