cleanflight/cleanflight #2885: Cleanflight v2.1.0-RC feedback.

hydra, 2017-07-10:
Please use this github issue for reporting _bugs/issues._
**Please use the slack support channel for support/help**.

alanfj, 2017-07-14:
Where can I download a recent Cleanflight PDF users manual for the latest Cleanflight version?

markgibbons25, 2017-07-16:
Does cleanflight support blheli32?

Is betaflight not going to be doing releases anymore since cleanflight is merging in the good and stable stuff regularly?

hydra, 2017-07-16:
@alanfj We'll get one prepared soon, the manual is somewhat out of date - unfortunately most of the contributors ignore documentation tasks.

hydra, 2017-07-16:
@markgibbons25 I need some blheli32 ESCs for testing.  If anyone can help with that drop me a message via the Cleanflight slack channel. http://cleanflight.com/slack - it will in due course as @4712 is a contributor.

bigd942, 2017-07-19:
Apologies, mix of feedback for Configurator and Firmware

Using a CL_Racing F4 board with Spedix 25a ESC (16.6) and T Motor F40 lll I had some motor twitches when set to Multishot, change over to DShot 600 and no issues so far. Had to set vbat_scale through the CLI as the tab on the configurator (2.1.0) was greyed out for me. For the OSD setup everthing worked ok (reals problems in betaflight 3.2 with this on this board), some of the buttons seemed temperamental/sluggish i.e turned off and the wouldn't stay off, reconnected and they worked, then didn't work. For the OSD setup I think it would be good to have everything not shown by default (Logo only maybe) or a button to turn them all off in one go, I only want rssi, voltage and flytime and it took a while turning everything off. Timers aren't clear on the osd selection tab, calling them Timer 1 and Timer 2 - possible change to Fly Min and On Min. The timer tab on the right hand side of the osd page is a bit confusing as well. Recommend a note near the OSD screen that the preview will not change font when you upload a new one, possible this can be fixed.

Mmarsian, 2017-07-23:
SP Racing OSD/PDB stopped working in conjunction with SP Racing F3 EVO (original hardware from SP). 2.0.5 was the last version that worked (with wrong current monitor values) . RC1 and RC2 don't show OSD output anymore, no font upload either (hangs the osd/pdb, no USB connect).

hydra, 2017-07-24:
@Mmarsian thanks for the feedback, I'm pretty sure I tested that locally here but I'll give it another test.  You enabled 'OSD' feature on the EVO right?

Mmarsian, 2017-07-24:
@hydra Thank you for your kind answer. I'll try to document the issue:
1. I flashed the SP Racing F3 EVO with SPRACINGF3EVO 2.1.0-RC2
2. I flashed the SP Racing F3 OSD/PDB with SPRACINGF3OSD v2.0.5stable
3. I configured OSD and used Fontmanager to load cleanflight font
4. I configured OSD on the EVO
5. it worked fine, see pictures:

[image]
[image]
[image]
[image]

Starting with stacked with EVO it even brings the Version 2.1.0 out working as a slave device
[image]

6. I flashed the OSD with SPRACINGFEOSD v2.1.0-RC2
[image]
7. When trying to upload font it hangs
[image]
8. The OSD screen is empty
9. Flashing the OSD back to v2.0.5stable, it works again.

I observed also:

in OSD Screen Mode is always AUTO, PAL or NTSC can be stored but is back on AUTO after reboot.

hydra, 2017-07-25:
@Mmarsian thanks for the detailed information, much appreciated.

hydra, 2017-07-25:
@Mmarsian please try the binary in https://github.com/cleanflight/cleanflight/pull/2904

Mmarsian, 2017-07-26:
@hydra I can make it short: it  works now! Thank you very much.

this one still remains, but I cannot see any bad impact in the functioning:

    in OSD Video Mode is always AUTO, PAL or NTSC can be stored but it's back to AUTO after reboot.
    unlike in your video UART1 cannot communicate with 230400 baud, it works ONLY with 115200.

amaussa, 2017-10-01:
my SPracingF3 work well with the version 2.03 after to update with the version 2.1 many things not work,
the GPS and the OSD, necessary to reinstall ?Thx

alanfj, 2017-10-01:
Try the latest Betaflight 3.17 version?

hydra, 2017-10-03:
please use rcgroups or the cleanflight slack channel for support.  this thread is for the release candidate feedback, a newer version exists now.

closed 2017-10-03 by hydra